OllisGit/OctoPrint-DisplayLayerProgress #5: Update intervall of the progress message

OllisGit, 2018-04-10:
At the moment the display/navbar is updated while the percentage is changed, but a better solution is to update the display if the layer is changed.
So the layers are "jumping" ;-(

First-Output: 1% 3/100
Second-Output: 2% 12/100

Problem: If I update the message while G-Code-Hook, it only works if Cura-Commands is found

Solution: ...not yet present...

TomW1605, 2018-04-11:
you could update it on both progress change and layer change so it will work for both

OllisGit, 2018-04-12:
The new Release 1.1.0 is out. Please test the new enhancement.
Thx, in advance
Olli

TomW1605, 2018-04-12:
this works well but i am not sure why you split it into 2 functions not just 1. i would also sagest that having it display the total lines (ether as 0/total or -/total) when you load the file would be better than when the print starts because that would then be in keeping with the behaviour of all the other stats displays already in octoprint.

OllisGit, 2018-04-12:
First point:  I am not sure how to implement it in one function, because the `g-code hook` needs to **return** the message layer command and update the navbar. In the `onPercentage hook` I need to send the message layer comand **directly** to the printer.
Maybe I missed something

Second point: I added this improvement to #12

TomW1605, 2018-04-12:
honestly i just wonted the layers (as the progress is already in the sidebar so i just moved the whole thing to its own function and called that from every other function. have a look at my fork (give it a bit i have just updated it with yours so i need to put my changes back)

OllisGit, 2018-04-12:
@TomW1605 Question about total file linenumber: Why is this information necessary? Or do you mean layers? If layers, then you already fixed this issue ;-)

TomW1605, 2018-04-13:
although i cant remember where i said that i probably meant layers

OllisGit, 2018-04-13:
The new Release 1.1.1 is out. Please test the new enhancement.
Thx, in advance
Olli

closed 2018-04-13 by OllisGit