OllisGit/OctoPrint-DisplayLayerProgress #10: S3D Support

nucu, 2018-04-11:
Simplify3D also adds layer comments in the gcode.

example:   
`; layer 10, Z = 1.640`

mikebzh44, 2018-04-12:
Hello OllisGit.

I'm testing a modded version of your plugin to read GCode generated by Simplify3D ;)

If this modded version work well, I will send you the code.

BTW, GCode created by S3D have layer comment, as shown in Nucu comment, but don't specify number of layers, like Cura. You need to count layers from GCode file. And this is also a line at the end the file with : 

`; layer end`

I will also check GCode file when you print more than 1 piece with dedicated process : 

https://www.simplify3d.com/support/articles/multi-part-printing/

OllisGit, 2018-04-12:
ahh... I was a litle bit faster than you ;-) because me implementation is done and I am now testing different G-Code Versions.
Also the plugin doesn't use the total_count any more, instead it counts the layer.

Hopfully I can release the new version end of the day.

OllisGit, 2018-04-12:
The new Release 1.1.0 is out. Please test the new enhancement.
Thx, in advance
Olli

TomW1605, 2018-04-12:
i have tested S3D style comments and they work

OllisGit, 2018-04-12:
Great!

closed 2018-04-12 by OllisGit

mikebzh44, 2018-04-12:
Good news !! I will test this release after my office day ;)

mikebzh44, 2018-04-12:
Work almost fine with a GCode file for printing 3 pieces, one by one ;)

`Progress: 73% Layer: 282/406`

Butif 406 layers = 100%, 282 layers should be 69% or 70%.

You can donwload the file here : https://www.dropbox.com/s/jpiso26hruh4mps/Lack_Angle_x3.gcode?dl=0

OllisGit, 2018-04-12:
hmm... I use the percentage value from Octoprint. So it is the same value as it used in the left blue progressbar.
I think the percentage is calculated on filesize. 

I will remove the percentage from the navbar in the next release, because it was redundant.

If you need the layer-percentage instead of the "normal" percentage, please so kind and create a new issue.

Thx, in advance
Olli