OllisGit/OctoPrint-DisplayLayerProgress #8: add support for getting total layer number from last layer change comment

TomW1605, 2018-04-11:
this would make it very easy for people to use your plugin with slic3r as well. slic3r allows for custom gcode to be added at each layer change and has a [layer_num] placeholder that lets you duplicate the `;LAYER:0` format from cura. however, slic3r has nothing that gives the total layers. it is possible with post processing but if you could use the last layer change comment then this would not be necessary.

OllisGit, 2018-04-11:
just to make sure that I understand your enhancment:
- instead of using the `;LAYER_COUNT:` the plugin should use the last `;LAYER:` comment
- slic3r could provide the Cura-Layer format (`;LAYER:`), right!?!

I think this is a realy good enhancement and it should be easy to implement.
I will start implementing this ;-)

TomW1605, 2018-04-12:
yes it can. under printer settings there are custom gcode options
[image]

OllisGit, 2018-04-12:
The new Release 1.1.0 is out. Please test the new enhancement.
Thx, in advance
Olli

TomW1605, 2018-04-12:
it works for me

OllisGit, 2018-04-12:
Great!

closed 2018-04-12 by OllisGit

reloxx13, 2019-01-19:
for lazy guys cuz in the comment its an image

`;LAYER:[layer_num]`

would be nice to have it as default, just trying out slic3r and now the print is without layer count :S

vitt82, 2020-05-07:
> for the lazy copy&paste guys cuz in the comment above its an image
> 
> `;LAYER:[layer_num]`
> 
> would be nice to have it as default, just trying out slic3r and now the print is without layer count :S

🥇

Doprintityourself, 2020-12-17:
Hi It is not working for me here
Prusa Slicer 2.2 and 2.3Beta3
I have 
;LAYER:[layer_num]
in bevore layer change gcode
[image]
but still getting no layer indicator as a warning 
[image]
the gcode file contains the wanted line
[image]

but I only see
[image]
and in top status line layer of is missing
[image]

I have uploaded the file new but not working

Interestingly the file should be changed during the upload!?
but it seems totaly the same after download the upload and comparison of the two files
Should the uploaded file be modified?
thanks in advance

Doprintityourself, 2020-12-17:
Very strange, reset all settings, rebooted and uploaded again, now it seems to work partially
Layer is progressing but total height is still not displayed 
[image]

Doprintityourself, 2020-12-17:
what the hack is going on after finishing it displays higher than total
[image]

I will now totally uninstall clean data and do a fresh install

Doprintityourself, 2020-12-17:
closed seems to work again after wiping data and install an reinstall plugin

Doprintityourself, 2020-12-17:
Not consistant 
I have installed it on my seccon server fresh, never had this plugin installed before
I have to say that it is a multiple instance install!
Outcome:
first printer 
not working properly 
[image]
seccond printer on xxx.xxx.xxx:5001
as it should be
[image]

any idea could the plugin fail becaus of mutliple instances?

sandboxgangster, 2021-06-07:
> ;LAYER:[layer_num]

Simple Thing your must add a SPACE after ; (; LAYER:[layer_num])