OllisGit/OctoPrint-DisplayLayerProgress #9: add option for putting in sidebar instead

TomW1605, 2018-04-11:
this would be quite useful particularly to people who already use plugins that add to the nav bar. it also makes more sense for the remaining layers to be with the progress bar, data printed counter and time remaining estimate as these are all print progress related.

TomW1605, 2018-04-12:
[image]
this is the problem with having it in the navbar

TomW1605, 2018-04-14:
unless it is easy to make the position in the sidebar customisable, i think the best place would be just above the progress bar

OllisGit, 2018-04-14:
The new Release 1.2.0 is out. Please test the enhancement.
Thx, in advance
Olli

closed 2018-04-14 by OllisGit