OllisGit/OctoPrint-DisplayLayerProgress #12: More descriptive output.

jprochazka, 2018-04-12:
Just a thought but not necessary by no means.
Got a lot of room there maybe a more descriptive status line such as:
`25% Printing layer X of X...`

OllisGit, 2018-04-12:
Thx, for your feedback!
Yes, the statusline is really "naked". I need to optimize this in the near future.

At the moment I didn't use the "M73 Progress-Plugin", but maybe I can adapt this feature as well.

OllisGit, 2018-04-12:
Also a hint from @TomW1605  in #5 : 

> i would also sagest that having it display the total lines (ether as 0/total or -/total) when you load the file would be better than when the print starts because that would then be in keeping with the behaviour of all the other stats displays already in octoprint.

OllisGit, 2018-04-14:
The new Release 1.2.0 is out. Please test the enhancement.
@jprochazka  for the "M73 Progress-Plugin" I will create a new issue

Thx, in advance
Olli

closed 2018-04-14 by OllisGit

OllisGit, 2018-04-15:
@jprochazka see #20 please test it!