OllisGit/OctoPrint-DisplayLayerProgress #14: add settings page

TomW1605, 2018-04-14:
i think a settings page that lets you chose what to display (progress/layers/both) on the printer and navbar separately would be good. also when available put an option here to move it from navbar to sidebar.

TomW1605, 2018-04-14:
it could even allow custom formatting with placeholders for progress (`[progress]`), curent layer (`[layer]`) and max layer (`[max_layer]`)

OllisGit, 2018-04-14:
Yes, I am implementing this right now:
Location-Option and what to output.

TomW1605, 2018-04-14:
i just noticed that your solution to the underscore issue was to check if it your printer and only add it if it is. this is good for now but this would be something else worth adding to the settings.

TomW1605, 2018-04-14:
on further reflection i think `[PROGRESS]`, `[CURRENT_LAYER]` and `[MAX_LAYER]` would be better placeholders

OllisGit, 2018-04-14:
Good point! 
I will add this to the "setting-page" enhancement.

OllisGit, 2018-04-14:
The new Release 1.2.0 is out. Please test the enhancement.
Thx, in advance
Olli

closed 2018-04-14 by OllisGit