OllisGit/OctoPrint-DisplayLayerProgress #18: AnyCubic I3 Mega

XenZen, 2018-04-14:
Progress is shown on OctoPrint Page ("Layer 5 /5") but printer screen shows only: Progress: --

OllisGit, 2018-04-14:
Which version of the Plugin do you use?
A couple of minutes ago I released the Version 1.2.0 with a new option to define the Layer-Message.
Please try the new version.
If it is not working, please try sending the following M117-comand while printing via Terminal:

`M117 99% 12 of 210`

The M117 command could only be generated by the plugin, if layer-comments could be found after uploading (see startpage)

Download the g-code directly from OctoPrint and check if there are M117 commands.

BR
Olli

XenZen, 2018-04-15:
the printer responds with 'ok' for
M117 99% 12 of 210
but doesnt display progress on its screen.

OllisGit, 2018-04-15:
hmm..strange...maybe the firmware does not recognize the percent-character.
Please send the command again without percent.
If it help, you can change the output in the settings,
BR
Olli

XenZen, 2018-04-18:
seems that firmware doesnt display m117. without percent char or only one number without "a of b" part.
anyway its usefull to see layer number on OctoPrint (:

OllisGit, 2018-04-20:
Hi, I did some research and you are right: TheAnyCubic doesn't support M117...but you can "upgrade" to Marlin ;-)
See https://makerware.thingiverse.com/groups/anycubic-i3-mega/forums/general/topic:25471
And https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT

closed 2018-04-20 by OllisGit

djalexz, 2018-12-22:
Hi, any Chance to Show the Status messages on anycubic i3 mega Display? I have installed Marlin but the M117 command does not work. Any idea? Thank you

OllisGit, 2018-12-22:
What kind of response do you receive after you send the M117 command via terminal to the Printer?
If M117 is not working please contact the marlin community!

renegade2k, 2022-01-17:
Well ... As this question is only closed, because of no reaction and not because of it was solved i wanted to give a final word to that:
i got the same problem ... anycubic i3 mega + marlin firmware and was wondering, why i got no display on the printer.
The anycubic printer, or to be more precise: the board + it's display, **DOES NOT SUPPORT M117 AT ALL**.
And you cannot fix this by another firmware like marlin.
It's simply impossible to send non-standard massages to this display.
That's it!
As far as i got, there is no workaround, no fix and no custom firmware to make it possible.