OllisGit/OctoPrint-DisplayLayerProgress #19: change the profile input to a checkbox

TomW1605, 2018-04-15:
[image]
this would be much better as a checkbox. something like this
[image]

TomW1605, 2018-04-15:
also i think this is a better format for the labels in the top section
[image]

OllisGit, 2018-04-15:
I choose the profile-name approach, because I had some profile dependent usecases in mind and maybe I will implement it in future.

Because these usecases are far away I will choose your approach and switch to a checkbox. 

Thx, for your ideas
Olli

TomW1605, 2018-04-15:
if you are going to go tith profile names then you should make it a list so people can add more than one profile

OllisGit, 2018-04-15:
Exactly, a list is much better!
By the way: what kind of Printer(s) do you use?

TomW1605, 2018-04-15:
it is a diy kit based off the prusa I3 and using the same Marlin firmware (note it does not have a display although i have been thinking of adding one for a while)