OllisGit/OctoPrint-DisplayLayerProgress #22: Layers not displayed

JanSykor, 2018-04-16:
Hello
I have installed version 1.2.1 of your addon on last version of Octoprint 1.3.8 on octopi 0.13
Printer is Builder Dual feed
On Octoprint and printer is displayed only percentage
File was sliced by Simplify (Cura also not working) and inside is layer information in this format:
; layer 43, Z = 4.29
On printer display is message like:
45% - of 
Please see attachments bellow.
Modification of message working for NavBar good. I s it possible use this setting also for Status panel?
Could you help me please, where I have problem? I think it can be very usefull.
It can be problem in octopi version 0.13?

Thanks 
Jan
[Layer Progress.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1914589/Layer.Progress.zip)

OllisGit, 2018-04-16:
First Point) hmm.. I put your delivered G-Code to my printer (local-storage, NOT SD-Card) and the total layer-information of 911 was shown.
Do you stored the gcode on sd-card? Maybe the file preprocessor didn't work on sd-card. I never tested it, because I have no card.

Second Point) From my point of view the output for the statebar should be fixed. Only support of multiple languages. At the moment English, German. I can add Czech. Layer -> Vrstva

[image]

JanSykor, 2018-04-17:
It is strange.
I´m saving gcodes to local storage (SD of raspberry). Anyway problem is not in gcode, but it is somewhere in my system.
I have this installed on 2 systems (Raspbian and Octopi 0.13) and both have similar behaviour. On next prints addon working but showing different value then Gcode viewer. Usualy +1 layer on actual and +1 layer on total quantity of layers then Gcode viewer. But sometime that differencies was higher.
[image]

If adding of other language is easy for you - it is good for me also. If you are busy - you can keep english.
Thanks for your help.
I found in langauge file only 2 strings - its correct? There is no settings strings.

```
# Czech translations for DisplayLayerProgress.
# Copyright (C) 2018 The OctoPrint Project
# This file is distributed under the same license as the
# DisplayLayerProgress project.
# FIRST AUTHOR <EMAIL@ADDRESS>, 2018.
#
msgid ""
msgstr ""
"Project-Id-Version: DisplayLayerProgress 1.1.1\n"
"Report-Msgid-Bugs-To: i18n@octoprint.org\n"
"POT-Creation-Date: 2018-04-14 11:45+0200\n"
"PO-Revision-Date: 2018-04-14 11:42+0200\n"
"Last-Translator: FULL NAME <EMAIL@ADDRESS>\n"
"Language: cs\n"
"Language-Team: cz <jan.sykora.8108@seznam.cz>\n"
"Plural-Forms: nplurals=2; plural=(n != 1)\n"
"MIME-Version: 1.0\n"
"Content-Type: text/plain; charset=utf-8\n"
"Content-Transfer-Encoding: 8bit\n"
"Generated-By: Babel 2.5.1\n"

#: octoprint_DisplayLayerProgress/static/js/DisplayLayerProgress.js:25
msgid "Layer"
msgstr "Vrstva"

#: octoprint_DisplayLayerProgress/static/js/DisplayLayerProgress.js:26
msgid "Shows the layer information"
msgstr "Zobrazí informace o vrstvě"
```

OllisGit, 2018-04-17:
1) different layer count: yes, that is a point. Some slicer starts there layer-comments with 0 (Cura) other starts 1. So that is the explanation about differnt.
But I didn't check, how the gcode view handles the layers via javascript.
I will crate a new issue about this bug.

2) yes, at moment only statebar-messages are I18N ready and not the settings.
I will put your above language file to the next release and I will post here the translation of the settings, so hopfully you can translate it to Czech.

JanSykor, 2018-04-17:
Its my pleasure to help you with translation to Czech. Unfortunately it's only this one way which I can help to you.

OllisGit, 2018-04-17:
Hi @JanSykor here is the new translation file for the settings-page please fill the msgstr segment.
Thx, in advance

```
#: octoprint_DisplayLayerProgress/static/js/DisplayLayerProgress.js:23
msgid "Layer"
msgstr "Vrstva"

#: octoprint_DisplayLayerProgress/static/js/DisplayLayerProgress.js:24
#, fuzzy
msgid "Shows the layer information"
msgstr "Zobrazí informace o vrstvě"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:3
msgid "Visibility"
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:6
msgid ""
"You can define where you want to display the layer- and progress-"
"information and also how the message should be shown."
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:9
msgid ""
"Use the following keywords as a placeholder:<code>[current_layer], "
"[total_layers], [progress]</code>"
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:13
msgid "Navigationbar"
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:20
msgid "Printer Display"
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:29
msgid "Display Workaround for ANET E10"
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:32
msgid "Add trailing character"
msgstr ""

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:36
msgid ""
"Use this if your printer display cuts off the last character of the layer"
" message."
msgstr ""
```

JanSykor, 2018-04-18:
Thanks.
Here is Czech version
Jan

```
#: octoprint_DisplayLayerProgress/static/js/DisplayLayerProgress.js:23
msgid "Layer"
msgstr "Vrstva"

#: octoprint_DisplayLayerProgress/static/js/DisplayLayerProgress.js:24
#, fuzzy
msgid "Shows the layer information"
msgstr "Zobrazí informace o vrstvě"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:3
msgid "Visibility"
msgstr "Nastavení zobrazení"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:6
msgid ""
"You can define where you want to display the layer- and progress-"
"information and also how the message should be shown."
msgstr ""
"Zde můžete nastavit kde chcete zobrazit informace o vrstvách"
" a stavu tisku a jak má tato zpráva vypadat."

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:9
msgid ""
"Use the following keywords as a placeholder:<code>[current_layer], "
"[total_layers], [progress]</code>"
msgstr ""
"Následující klíčová slova použijte jako zástupný symbol: <code>[current_layer], "
"[total_layers], [progress]</code>"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:13
msgid "Navigationbar"
msgstr "Navigační (horní) lišta"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:20
msgid "Printer Display"
msgstr "Displej tiskárny"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:29
msgid "Display Workaround for ANET E10"
msgstr "Úprava zobrazení pro ANET E10"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:32
msgid "Add trailing character"
msgstr "Přidat chybějící znak"

#: octoprint_DisplayLayerProgress/templates/DisplayLayerProgress_settings.jinja2:36
msgid ""
"Use this if your printer display cuts off the last character of the layer"
" message."
msgstr ""
"Použij toto pokud displej tiskárny nezobrazí poslední znak zprávy"
" o vrstvě." 
```

OllisGit, 2018-04-19:
The new Release 1.2.2 is out. Please test the translation.
Thx, in advance
Olli

closed 2018-04-19 by OllisGit

JanSykor, 2018-04-20:
Thanks @OllisGit 
I´m already installed update, but from unknown reason all octoprint language update I can see next day. Reloading of pages not change language same day.
I will inform you later.
best regards
Jan

JanSykor, 2018-04-22:
Hello @OllisGit
After few day I can confirm that language was change in Status side bar, **but not in the setting menu**. There is still in english. Your switching to german language is working?
Do you have some idea what I´m doing wrong?
Thanks
Jan

OllisGit, 2018-04-22:
Hi @JanSykor ,
I created a new issue (#27), because this one is closed and I don't monitor closed issues.

Do you have other plugins which support Czech language in settings-menu, so that I can check what they are doing differently? Please report this information to #27 

Thx,
Olli