OllisGit/OctoPrint-DisplayLayerProgress #23: Total Layercount doesn't match wich gcode-viewer

OllisGit, 2018-04-17:
1) Different slicer use different layer-numbering. E.g. Cura starts at layer 0. 
Thats the reason that at the moment the plugin adds plus one.

2) Check how the javascript base gcode viewer works, to evaluate the layer hight

OllisGit, 2018-04-17:
puhh.. I looked into the gcode javascript...it is very complicated (don't get me wrong, javascript is good designed) It is a fully gcode parser.

It is not an easy task to find a good strategie for evaluating the layer count.
The current strategie (V1.2.x): use the layer comments -> really simple
Another approach could be to check each Z-movement (unique movement needed. If you print many objects one after the other (e.g. four towers in each corner)).

conclusion at the moment: remove the addition +1. "trust" the layer comment value!

Any opinions on this topic are welcome!

BR
Olli

JanSykor, 2018-04-18:
I think addition of +1 is not big problem, but on bottom picture I have diff +5 (-1) and I don´t understand why. Gcode included in zip file - this was printed all models together.
Best regards
Jan

[NECK_BASE-f3.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1923502/NECK_BASE-f3.zip)

[image]

JanSykor, 2018-04-18:
Here is some other gcodes:
Drzak opto cidla na Alu profil.gcode - only one model - layers OK - same as Gcode viewer

2018_olepování AL bezelu_65FX700_podložka_BTM.gcode - layers +1 - printing sequentialy one by one
Both sliced by Cura for same printer
I hope that will help
Jan
[gcodes.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1923574/gcodes.zip)

JanSykor, 2018-04-19:
Here is file sliced in simplify for Builder - result was +1 in actual and +1 in total layer
Jan

[ukazovátko.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1926961/ukazovatko.zip)

richgain, 2018-04-27:
How about adding a config option to allow users to manually adjust for their set up?
My printer consistently reports Layer: 2/XX when it's printing the first layer!

richgain, 2018-04-27:
Hi Olli
Thanks for the advice on helping to understand the problem.
Just to clarify, my display doesn't actually say "2/XX", I simply meant that it always starts at 2. The other number (the maximum layer) is displaying correctly. The file I am currently printing says 2/179 on the first layer.
I have just done as you suggested and compared the first and last layer message as they come out of S3D and again after downloading the file from Octoprint.

Before:
-  First  -  ; layer 1, Z = 0.100
-  Last  -  ; layer 178, Z = 17.800
After:
-  First  -  M117 INDICATOR-Layer1
-  Last  -  M117 INDICATOR-Layer178

So the problem is not with the gcode, but with the display.

OllisGit, 2018-04-27:
ahhh, okay, now I understand. ....damn that takes a long time that this issue reach my brain ;-) 

So the problem is the +1 I added, because of the CURA behaviour which start at layer 0.

In the next Version (V1.5) I will provide a "layer-adjustment-option".

BR
Olli

closed 2018-05-05 by OllisGit