OllisGit/OctoPrint-DisplayLayerProgress #29: Layer counter shows only - / -

beejayf, 2018-04-23:
Layer current and layer total does not show on CR-10 Marlin 1.1.8

OllisGit, 2018-04-23:
Did you already read the information of the start page?

> * The layer information output only works with "Cura"/"Simplify3D" generated G-Code, because these Slicer insert the layer information directly as comments in the file. If you want to use "slic3r", see Enhancement #8

beejayf, 2018-04-23:
I didn’t – thanks for the hint!

Von: OllisGit
Gesendet: Montag, 23. April 2018 22:47
An: OllisGit/OctoPrint-DisplayLayerProgress
Cc: beejayf; Author
Betreff: Re: [OllisGit/OctoPrint-DisplayLayerProgress] Layer counter showsonly - / - (#29)

Did you already read the information of the start page?
• The layer information output only works with "Cura"/"Simplify3D" generated G-Code, because these Slicer insert the layer information directly as comments in the file. If you want to use "slic3r", see Enhancement #8

OllisGit, 2018-04-23:
If you have further issues, please create a new one.
This issue is now closed.
BR
Olli

closed 2018-04-23 by OllisGit