OllisGit/OctoPrint-DisplayLayerProgress #30: Some info are not displayed (Anet A8 20 char LCD display)

mikebzh44, 2018-04-24:
Hi.

Thanks for this awesome plugin !!

Some informations are not displayed on my Anet A8 20 charaters LCD "stock" display.

For example, I have : 

`21% L: 35 of 389 H: `

You should use / instead of " of " so you can display height : 

`21% L: 35/389 H:12.4`

Pattern example : 

`xx% L:xxx/xxx H:xx.x` = 20 char ;) 

Thanks.

OllisGit, 2018-04-24:
I shorten the default printer-message  like you request....but it is also to long, if you want to display the total height.

Old (39 chars):
`xxx% L: xxx of xxx H: xxx.xx of xxx.xx`

New (30):
`xxx% L:xxx/xxx H:xxx.xx/xxx.xx`

So you must adjust the message pattern...or you can buy a new display (just kidding)

BR
Olli

mikebzh44, 2018-04-24:
I have already plan to change my "poor" LCD display ;)

Thanks for the update.

OllisGit, 2018-04-25:
@mikebzh44 You are the reason for that trouble today! JUST KIDDING!!!!
I adopted your suggested message pattern and some printers "crashed" when they receive a colon in there message.

Anyway, this how opensource works, feedback from users -> improve quality of product -> great!
I like it ;-)

mikebzh44, 2018-04-26:
Oops ;)

But I don't understand as your old message already had colons (':' is a colon, right ?) : 

Old (39 chars):
`xxx% L: xxx of xxx H: xxx.xx of xxx.xx`

OllisGit, 2018-04-26:
Yes, but the colon-pattern was only online for a few days. So my assumption is that this pattern wasn't used by "problematic-printers".

closed 2018-04-26 by OllisGit