OllisGit/OctoPrint-DisplayLayerProgress #31: Latest update causes print to fail, due to printer display format?

kennethpangky, 2018-04-25:
On the Ender-3, it worked fine until the most recent update.

Default printer display is : [progress]% L:[current_layer]/[total_layers] H:[current_height]/[total_height]
I have not changed it.

On starting a print, terminal shows the bed temp reaching required temp. When it's due for hotend to heat up, the plugin floods the printer with the below commands crashing the printer (terminal shows 1500 lines, too many to buffer when it was just 300+ right before the crash) - I can't cancel the print via Octoprint until I disconnect or power down the printer. It just gets stuck at the cancelling state.

`Recv: Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
Recv: ok
Send: N2 M117 0% L:-/20 H:0/15.00*34
Recv: Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
Recv: ok
Send: N2 M117 0% L:-/20 H:0/15.00*34
Recv: Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
Recv: ok`

Tried printing again with printer display disabled. Works fine. Saw from another issue that the printer display format was changed to accommodate the length, so I guess this is related?

OllisGit, 2018-04-25:
Sorry to hear that!
I changed the default printer message (more information -> longer text), but some printer ignores the overhanging characters and some stops printing. grrrr!

I shorten the default message and added a warning note to the settings page in the new release 1.3.3.

Please give it a try!

Thx, in advance
Olli

kennethpangky, 2018-04-25:
Hi, same issue.

Current default printer display format is : [progress]% L:[current_layer]/[total_layers]

Errors received as follows

`Send: N2 M117 0% L:-/146*116
Recv: Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
Recv: ok
Send: N2 M117 0% L:-/146*116
Recv: Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
Recv: ok
Send: N2 M117 0% L:-/146*116
Recv: Error:No Checksum with line number, Last Line: 1
Recv: Resend: 2
Recv: ok`

OllisGit, 2018-04-25:
hmm... I also changed the word 'of' with '/'. Maybe thats the issue. Try to replace 'special characters'  like / % :

Please send following gcodes via terminal to your printer:

`M117 0% L:-/146`

`M117 0% L:- 146`

OllisGit, 2018-04-25:
I had the opertunity to test the plugin with a different printer and found out the reason for this strange behaviour:
The problem is the new default printer-message. There is a : (colon) in the message.
After sending this faulty command the printer runs into a kind of endless loop, see terminal tab.

Remove the colon (or replace by =) or update to a the new release V1.3.4. 

Please test it
Olli

kennethpangky, 2018-04-26:
Ok, it works now! Your support is much appreciated - you could have easily told me to try out other formats that would work (not complicated at all), but you took the effort to narrow down the specific keystroke that cause the problem.

It appears it's a printer specific problem. Those kinds are always a pain to fix. 😵

OllisGit, 2018-04-26:
Yes, thats the point. As a Plugin-Developer you need a lot of differnt Printer-Setups to tests "everthing".
..or you use this german-pattern "Bananen Software: reift beim Kunden" 
possible translation: "Banana Software: incubate/grows after delivery"

...or you have a big, frindly and stress-resistant community ;-)

closed 2018-04-27 by OllisGit