OllisGit/OctoPrint-DisplayLayerProgress #32: V1.3.2 breaks Z-Movement on ANET A8

fjibril, 2018-04-25:
Took me some time to figure out it was this plugin. Printing from SD works fine, but the G-Code for Z-Axis Movement from Octoprint simply does nothing. Not from printing, not from manual Control in Octoprint, and not when sending the G-Code from Terminal Tab. No further error messages in Logs. Just no reaction. Homing Z-Axis works, though. Different G-Code. Sending X- and Y- Axis movement G-Code works via all above mentioned methods.

I just installed this plugin and left config to default. Update to 1.3.2 broke it. Disabled Plugin, Z-Axis works again.

OllisGit, 2018-04-25:
Sorry to hear that!
I changed the default printer message (more information -> longer text), but some printer ignores the overhanging characters and some stops printing. grrrr!

I shorten the default message and added a warning note to the settings page in the new release 1.3.3.

Please give it a try!

Thx, in advance
Olli

fjibril, 2018-04-25:
Seems to have fixed it, yes. Thank you!

closed 2018-04-25 by fjibril