OllisGit/OctoPrint-DisplayLayerProgress #33: Layer count always "- / -"

TheBlueGreen, 2018-04-25:
I've been using this plugin for a week or so and have updated several times but my LCD and top bar in the browser never show a layer count. The percentage works fine but it will display as "99% - / -" (99% is an example but the "-"s should be the layer counts). I'm using an Original Prusa MK3 on the latest OctoPrint and latest Prusa firmware. I don't believe it is a problem with my printer as the layer counts never worked in the OctoPrint UI either. Any ideas? Thanks

OllisGit, 2018-04-25:
Which slicer do you use?

Please look at the startpage for the description how the plugin works.
Short: the g-code must contain layer information via predefined comment-format

BR
Olli

richgain, 2018-04-27:
I had the same issue until I realised that my printer profile in Simplify3D  had an entry in the Post Processing script box that was automatically changing the layer message.
Once I deleted that script, layer counts started displaying correctly.

ejjenkins, 2018-04-28:
Mine does the same thing, I use Simplify3D as well, nothing is in the PostProcessing script box.  Here is the G-Code:
[BC_Wheel.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1957117/BC_Wheel.zip)

OllisGit, 2018-04-28:
hmmm... I put your gcode to my octoprint and after selecting it shows "Layer: - / 49" in the statusbar.

Which plugin version do you use?
Yesterday I released Version 1.4.1. Bugfix for "layer-patterns" without a number (see #36).
Maybe you can test the new one.

If it's not working, please also provide a downloaded version of your g-code (upload to octoprint, download from octoprint). Maybe the layer-indicator is not correctly created.

BR
Olli

OllisGit, 2018-04-29:
Thats the point!! I reproduced the behavior.
The file preProcessor is not working if file was just copied to upload-folder, you need to use:
a) web-page via drag-drop
b) rest-api (e.g. like the slicer do)

Why is uploading not working for you, what is the problem?

ejjenkins, 2018-04-29:
The uploading is not a problem, saving directly from the slicer is more convenient.  At first I did not realize that your plugin parsed the file during upload.  Now that I do, no problem.

Thank you.

OllisGit, 2018-04-29:
No Problem!
I close this issue, if you have other questions/bugs, please creae a new one!

BR
Olli

closed 2018-04-29 by OllisGit