OllisGit/OctoPrint-DisplayLayerProgress #34: After installing DisplayLayerProgress, z-up never works.

leftside2, 2018-04-25:
Octoprint 1.3.8, Marlin on Anet A8. 
After removing the plugin everything works fine.

OllisGit, 2018-04-25:
Sorry to hear that!
Each time the z-axis is moved, the display is updated now. This "feature" was added in Version 1.3.0

What kind of Raspberry do you use (Raspi2 or 3)?
Maybe the regular-expression I use is to "much" for the cpu.

Conclusion: Downgrade to Version 1.2.5 or wait for a new release!

OllisGit, 2018-04-25:
I had the opertunity to test the plugin with a different printer and found out the reason for this strange behaviour:
The problem is the new default printer-message. There is a : (colon) in the message.
After sending this faulty command the printer run into a kind of endless loop, see terminal tab.

Remove the colon (replace by =) or update to a the new releasee V1.3.4. 

Please test it
Olli

bookledge, 2018-04-26:
I have this issue in v1.3.2 .    after v1.3.3 update, the problem has been solved. 
thanks..

leftside2, 2018-08-16:
Thanks, man!