OllisGit/OctoPrint-DisplayLayerProgress #35: Problems after updating to v1.3.3

phcoetzee, 2018-04-25:
After updating plugin from v1.3.0 to 1.3.3 my octoprint becomes unresponsive. When I click on a gcode to start printing, nothing happens. No rise in bed temp or nozzle temp and even if I want to cancel the print, nothing happens. I have to reboot my octoprint. If I disable the plugin, everything works fine. With v1.3.0 everything was also working fine. Running octoprint v 1.3.8.

OllisGit, 2018-04-25:
I had the opertunity to test the plugin with a different printer and found out the reason for this strange behaviour:
The problem is the new default printer-message. There is a : (colon) in the message.
After sending this faulty command the printer run into a kind of endless loop, see terminal tab.

Remove the colon (replace by =) or update to a the new releasee V1.3.4. 

Please test it
Olli

phcoetzee, 2018-04-26:
Hi Olli,

Looks like the bug has been fixed now with v1.3.4

Thank you very much

Pieter Coetzee

closed 2018-04-26 by phcoetzee