OllisGit/OctoPrint-DisplayLayerProgress #37: Multiple M117 are sent, while there is no change in layers or percents.

TheMadKow, 2018-04-25:
[image]

I'm using M117 popup display, so it's easy to see.
After updating to new version, it keeps sending M117 messages (couple every second I guess), while layer count and percentage of print stays the same. It goes forever 😀 

I don't know if it affects print speed or anything. Or is it supposed to work that way, but it wasn't the case before latest update (1.3.3)

TheMadKow, 2018-04-25:
Also, I think it might be whenever the print head z-hops a bit.

richgain, 2018-04-27:
+1
I noticed the same thing yesterday. Printing 10 parts on the plate and every Z hop sends a new message.
This makes it virtually impossible to control the printer via Nautilus.

OllisGit, 2018-04-27:
I don't know exactly what your "usecase" for this M117 popup is, but is it necessary to bring up a single popup for each M117 message?
An improvement could be to show a single popup like a termial/textarea where the lastes message is on top. Maybe someon suggested this to the plugin-author.

By the way, I think  I will finish the mentioned improvment this weekend.

BR
Olli

OllisGit, 2018-04-27:
In the new release V1.3.5 the amount of sending M117 commands to the printer is improved.

Sending depends now on what keywords do you use in the printer message pattern (see plugin-settings)
E.g. if you only use `progress`, then each time the progress is changed the M117 command is send or
if you use height-keyword(s), the M117 message is send when the height is changed.

Please test the new version.
Olli

OllisGit, 2018-04-27:
I added in Version 1.4 a "Desktop Printer-Display". The PopUp shows all M117 messages.
Maybe it is something you could use.
[image]

closed 2018-04-29 by OllisGit