OllisGit/OctoPrint-DisplayLayerProgress #38: Last Digit is missing in max Layer Number

Fiiti, 2018-05-02:
If there are e.g. 92 Layer to print, I get in the Printer Display (marlin FW 1.0) only e.g. shown "31% L=19/9".
But if would the total amount of layer e.g. 251, the display will show e. g. "11% L=31/25".

At the very end (after finishing the job) the display will show "100% L=92/92" or "100% L=251/251".
All Numbers are only examples.

So it seems, that every time the last digit is missing.
The Terminal has theese commandos:
Send: N29086 M117 56% L=48/99*85
So I wonder for what the *85 would be needed? Because these values are changing:
Send: N28880 M117 55% L=48/99*89
Send: N29314 M117 56% L=49/99*92

In my tests I sended M117 Commandos by my self and there was unfortunately also the last digit missing. I do not know, if this is an octopi or an marlin issue. But if i would add e. g. an . (dot) or any other character (except an space), all was shown.
I wonder, because this issue is new for me. I printed many many prints with a lot of other slicer or also with octopi - and never before i had a missing character at the end.

So what is your opinion in this case?

Fi

Fiiti, 2018-05-02:
sorry - i am a dude. I found the settings and there you have already fixed this issue. Sorry man.

closed 2018-05-02 by Fiiti

user012flo, 2019-05-03:
hello fiiti,

can you explain me how to fixed your problem?
I have the same problem.