OllisGit/OctoPrint-DisplayLayerProgress #39: Enhancement: Add support for Simpliy3D and KISSlicer

b-morgan, 2018-05-05:
Please consider adding support for a couple more slicers. I have attached the gcode files produced when slicing 3DBenchy with CuraLE, Simplify3D, and KISSlicer. The CuraLE and Simplify3D files are for dual extruder (I can re-slice as single extruder if you'd prefer), the KISSlicer is for single extruder. All have layer number comments in varying formats. Only CuraLE has a layer count at the beginning. Let me know if you need anything else.

[files.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1976754/files.zip)

OllisGit, 2018-05-08:
@b-morgan 
Thanks for the files, it is really helpfull!
In the next version there will be an option to define custom layer-expressions. 
Also support for slicer without layer-numbers like KISSlicer (only layer-marker)

You use a lot of slicer, do you already test http://shapeforge.loria.fr/icesl/ ?
I looked at some youtube videos and it looks promising with nice features.

BR
Ollli

b-morgan, 2018-05-08:
I've looked at IceSL but haven't played with it much. When I first got my LulzBot TAZ 6, the Cura LulzBot Edition was pretty old (21.08). I wasn't completely happy with it so I started looking at other slicers. Cura from Ultimaker was at 2.x and I read a lot about Simplify3D. I also looked at slic3r and others. I was recently pointed to KISSlicer and it produces seems to work pretty well. I have a Dual Extruder and I haven't printed anything sliced with KISSlicer for the Dual Extruder (you need the PRO version but it wasn't that expensive).

OllisGit, 2018-05-08:
The new Release 1.6.0 is out. 
KISSlicer  is now included, also now you can define your own layer-comments-expressions.  

Please test the new enhancement.

Thx, in advance
Olli

closed 2018-05-11 by OllisGit