OllisGit/OctoPrint-DisplayLayerProgress #40: Conflict w/ Multi Colors

upD8R, 2018-05-05:
The plugin conflicts with default settings of the Multi Color Octoprint plugin. This plugin is also looking for the Layer comment field and fails after your plugin has processed the gCode file. I wonder why you replace the comment instead of adding the M117 command?

Anyway, changing the regular expression to "M117 INDICATOR-Layer{layer}" in Multi Colors did the job for me.

Thanks for your work, highly appreciated!

OllisGit, 2018-05-06:
Thx, for your feedback.
Because I want to be a "good citizen", so I don't want to "crash" other plugins, I will change the behaviour in the next release.

BR
Olli

upD8R, 2018-05-06:
Hi Olli,

maybe adding the RegExp above to your documentation is sufficient? It works but there is a dependency. I think a documentation change could be sufficient here.

Your choice 😜

Cheers, Mario

OllisGit, 2018-05-08:
The new Release 1.6.0 is out. 
The orig. comments are not droped anymore, also now you can define your own layer-comments-expressions.  See Plugin-Settings!

Please test the new enhancement.

Thx, in advance
Olli

closed 2018-05-11 by OllisGit