OllisGit/OctoPrint-DisplayLayerProgress #42: height offset

TomW1605, 2018-05-06:
every one of my models is consistently shown as .10mm taller than is is

TomW1605, 2018-05-07:
[image]
[image]
[small spool outside.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1981764/small.spool.outside.txt) just change the extention back to .gcode it wouldent let me upload with it

OllisGit, 2018-05-08:
I checked your provided gcode:

`Line 18912: G1 Z16.700 F9000.000`

This is the last Z-Movement in the file, so it looks correct for me.

TomW1605, 2018-05-08:
sorry the second screenshot there is was of the wrong thing. the gcode viewer in octoprint shows the height for that print as 16.60mm

OllisGit, 2018-05-08:
okay, the g-code viewer does a lot of "magic-calculation". I looked into the javascript code, but I couldn't understand everthing. So for me "magic" is best word for that.

I use the real values in the g-code!

TomW1605, 2018-05-08:
ok i have worked out the difrence. you are displaying the height of the nozel however they are displaying the height of the moddle itself. because i print with a .4mm first layer then ofset evrything but -.3mm i get the extra .1mm added to the nozel height. it is worth noting that evry online gcode viewer i can find gives the same height as the octoprint one (it is posable they are all using the same code though).

OllisGit, 2018-05-08:
thx, for the explanation. I didn't know that.
I will put this information into the documentation.

BR
Olli

closed 2018-05-11 by OllisGit