OllisGit/OctoPrint-DisplayLayerProgress #43: standard sidebar output

TomW1605, 2018-05-08:
i think that anything in the side bar should match the format and behavour (when files are loaded, deleated etc) of the entries that are there already. the 2 problems i have noticed is that the height shows 0 / 0mm instead of - / - when no file is selected (u can get this by selecting a file then deleating it) and the spinny loading thing should not be there.

OllisGit, 2018-05-08:
1) is fixed in the next release
2) the spinner is necessary, because if you have huge models > 20MB it takes a couple of minutes to parse on an "old" RaspberryPi. 
So a feedback to the user, that the height/layer information is currently evaluated is realy helpfull.

OllisGit, 2018-05-08:
The new release 1.6.0 is out. 
I changed the behaviour to -/-

Please test the new enhancement.

Thx, in advance
Olli

TomW1605, 2018-05-09:
better but now it is - / -mm and that just looks odd
[image]

OllisGit, 2018-05-09:
well...only a little bit...it is changed and in the next release you can see it...not see it.

BR Olli

OllisGit, 2018-05-11:
The new release 1.6.1 is out now, please test.

BR
Olli

closed 2018-05-11 by OllisGit