OllisGit/OctoPrint-DisplayLayerProgress #47: constantly says layer expression invalid but i have not changed it

TomW1605, 2018-05-10:


upD8R, 2018-05-11:
Same here. https://regex101.com/ also complains about the Cura line.
It seems I'm too stupid to understand RegExp.

upD8R, 2018-05-11:
Oh, it seem to work for me if I remove the RegExp for S3D and KISSlicer from the plugin settings. Anyway only using Cura so far.

OllisGit, 2018-05-11:
please put everthing from your plugin-settings rexex-textarea to your issue, so that I can analyse the expression/plugin.

I fixed a bug about the validation-state: If you press reset-settings, the old state of expression-validation is still there -> error-popup.

Please test V1.6.1

BR
Olli

upD8R, 2018-05-11:
Here's mine:
1 [;LAYER:([0-9]+).*] CURA
1 [; layer ([0-9]+),.*]	Simplify3D
count [; BEGIN_LAYER_OBJECT.*] KISSlicer

Will test 1.6.1 after my next print.

OllisGit, 2018-05-11:
hmmm...  looks ike the default. Couldn't find an error. Maybe some "hidden chars" like tab, return,...

upD8R, 2018-05-11:
1.6.1 installed, Reset to Defaults. Some changes reconfigured to match my setup.
RegExP as above.
I get the following error message while uploading the attached gCode.
**ATTENTION: DisplayProgressPlugin: LayerExpressions not valid! Check Plugin-Settings.**
[AI3M_Batterydisp_AAA.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/1994627/AI3M_Batterydisp_AAA.zip)

I deleted the uploaded file again, deleted the last 2 lines of RegExp. Uploaded again. It works.
I deleted the uploaded file again, added back the full 3 lines (copy & past from above) of RegExp. Uploaded again. It works. I checked the processed file.

So no final conclusion from my side ;-) A bit strange behavior, haven't checked any other file yet.

TomW1605, 2018-05-11:
mine acts exactaly the same way and also uses the default

OllisGit, 2018-05-11:
Thanks for the detailed description.
I found the bug. The save-function sends only the delta-values, but the plugin thought there is no expression in the textarea.

It's fixed in Version 1.6.2, please test.
BR
Olli

upD8R, 2018-05-15:
No issues so far with 1.6.2. Thanks Olli!

closed 2018-05-17 by OllisGit