OllisGit/OctoPrint-DisplayLayerProgress #48: DisplayLayerProgress with Octolapse

ukwebb, 2018-05-20:
while it works alone, DisplayLayerProgress has issues with the   height progress  when Octolapse is also enabled.
eg .... 
Progress: 18% Layer: 46 of 415 Height: 0.10 of 50.04mm

It always seems to show the zhop height in octolapse rather than the actual zheight

Can DisplayLayerProgress show the correct current layer height?

ukwebb, 2018-05-20:
the Octolapse developer suggests …. 
" So the display z plugin must not be taking the g90/g91 commands into consideration, or is not intercepting them. I wonder if this could be recreated with only gcode, or if it is specific to plugins that inject gcode."
maybe you guys could work together .. ? 
https://github.com/FormerLurker/Octolapse/issues/166#issuecomment-390507793

FormerLurker, 2018-05-20:
I'll try my hand at creating some gcode that will recreate the issue.  I should look at the code too probably :)

OllisGit, 2018-05-21:
the current height is captured from every z-movement. my plugin makes no differnce between absolut/relative z-movement.
if z-movement is send to printer, the plugin extract the z-value and send is to the display.
my assumption is, that the Octolapse-Plugin inject relative z-movements.
The current implementation use "queuing"-hook. see http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase
The next version of my plugin use "sent". Maybe this makes a difference.

A possible solution could be to detect absolute/relative commands (G90/G91) and indicate this in the display. I will think about it.

BR
Olli

ninjamojo, 2018-05-21:
You are correct.  Octolapse sends relative movements to control the Z axis.  Using the 'Sent' hook will prevent some race conditions but will not prevent the display of the relative movement.  I can't think of an easy way to display the current absolute z-height without a more complex position detection algorithm.

ukwebb, 2018-05-21:
it may not be this simple, but why not keep a running total ? 
something along the lines of … 
if home command height =0
if z height is set absolute then height = new zheight
if z height is relative then height = last height +/- relative distance
display height ...

OllisGit, 2018-05-22:
@ukwebb your are right...in theory it is just detecting the abs/rel-command and do some calculation.
But my concern is that the plugin needs to hold the state (the current height) and if it is out of sync the height is messed up.

But I will try to implement this behaviour in the next version (V1.8.0).

FormerLurker, 2018-05-29:
No problem!  I've been a bit swamped here (busy season) so it might take me a day or two, but I can't wait to try it out.  Thanks for your hard work on this.

OllisGit, 2018-05-29:
@FormerLurker Thx, take your time. 
The weather in north germany is to good to be in front of the keyboard..so my Octoprint-Activities are also on a "low-level" ;-)

FormerLurker, 2018-05-29:
@OllisGit, es is auch wunderschoen bei mir in Chicago :)

ukwebb, 2018-05-29:
running the same gcode, but with octolapse turned off results in …
Progress: 0% Layer: 0 of 246 Height: 0.24 of 49.44mm
Progress: 1% Layer: 1 of 246 Height: 0.44 of 49.44mm
Progress: 2% Layer: 2 of 246 Height: 0.64 of 49.44mm

OllisGit, 2018-05-30:
@ukwebb please provide your test gcode.

ukwebb, 2018-06-02:
do you mean the gcode file as created by cura, or the gcode as sent to the printer ie after octolapse has injected its new codes? - if its the latter, how do i capture that?

OllisGit, 2018-06-02:
You can't capture the octolapse gcode, because it is modified at runtime (I think). 
I want to use the inital g-code as a analysing starting point, but to be honest I am not sure what your problem is and what you expect.

Assumption: Octolaps move the extruder +0.1 and move it safely to the side. So, the Z-Movement is reflected correctly to the display.

Please give my a more detailed description of the issue.

ukwebb, 2018-06-02:
apologies for not explaining. everything done so far looks to be working great. the only thing its not doing is accounting for the initial zhop octolapse performs. 
from the same gcode file when octolapse is turned on the plugin says its 0.1mm (the octolapse zhop distance)  higher than it actually is (I've measured it and it is 0.24mm not 0.34 mm) . 
Turning octolapse off makes your plugin show the correct zheight.,

b-morgan, 2018-06-02:
Won't you get the same result if you set a Z-hop value in the slicer? That would take OctoLapse out of the equation.

ukwebb, 2018-06-02:
but why would the same gcode give two different layer heights depending on if octolapse is enabled or not?

b-morgan, 2018-06-02:
This plugin is capturing the Z values it sees in the gcode stream. Most of the gcode files generated by slicers with "default" parameters will start at a Z value close to zero and increase the Z value by increments of the layer height. This is the behavior you are seeing without OctoLapse active.

OctoLapse has the ability to "inject" gcode into the stream of gcode sent to the printer. This plugin also captures that gcode and it sees the zhop of .1mm OctoLapse injects so the nozzle will miss hitting the printed part while it moves the nozzle to a known position to take a picture. This plugin is just capturing Z values and assumes that Z is an always increasing value, so the zhop increases the layer height.

Most slicers have the ability to issue a zhop of their own when doing retractions as additional protection against unwanted interference with the printed part. Since this is an optional setting with a default of zero, this plugin behaves as you expect. If you were to turn on this option in your slicer and generate gcode, I believe this plugin will behave exactly the same as when OctoLapse is turned on.

Either this plugin displays the current value of Z including all the hops up and down or it has to have some way of differentiating the two sources of Z changes. The easiest way to do that would be to tell it what Z changes to ignore and what Z changes to process. Since the same gcode commands are used for both types of changes, the only thing left to use to tell the difference is the actual values of the changes (i.e process all changes of value "layer height" and ignore all changes of value "OctoLapse zhop" or "slicer zhop".

I believe this overly complicates this plugin and puts an undue burden on the user who would be responsible for setting these values for each and every print job.

[Marlin](http://marlinfw.org/meta/gcode/) has a "solution" to this problem in its implementation of G10, G11, M207, M208 but I don't believe there are any slicers that implement those at present and for OctoLapse to use them would make it Marlin specific which isn't a good idea either.

b-morgan, 2018-06-02:
On reflection, if this plugin were to display the current Z value instead of the highest Z value it has seen so far, it would display the "correct" Z value most of the time. 

The downside of this approach is that the value might appear "unstable" on the display. If the plugin were to keep track of the last positive and the last negative changes to the Z value, it might be able to "learn" which changes in Z value were for layer height and which were for Z hops.

ukwebb, 2018-07-08:
I understand what youre saying - basically it depends which order the plugins run to recognise whats happening. in that case - I guess 0.1mm out is way better that it was before :)

CDrawe, 2018-11-14:
I'm wondering if there Is a solution for the problem now?

FormerLurker, 2018-11-14:
As soon as I'm finished with my new plugin release I may take a look at the gcode sending hook to see if G91 can be easily taken into account, if @OllisGit approves.  I used relative z-movement in Octolapse because at the time I wasn't so confident in the position tracking capability, and moving Z +0.5mm and then -0.5mm (depending on lift height) for lift seemed a lot safer than moving to the expected height.  The same problem exists for XY tracking too, which is why Octolapse is so picky about start gcode (when using default settings).

You may be interested in looking at [this thread](https://discourse.octoprint.org/t/keeping-track-of-position-and-motor-mode-s/4358/13) over at the OctoPrint discourse site.  I'm considering separating the position tracking from Octolapse so that other plugins can take advantage.  It needs some work and has some known issues, but it correctly handles G90/G91, M80/M83, G92 (in most cases), G2/G3(alpha support) and others.  Once finished, this plugin could just query the position object for the current height.

If we're lucky a [new feature](https://github.com/MarlinFirmware/Marlin/issues/12192) may be coming to Marlin to help with this issue by giving us developers access to some currently read-only firmware data (specifically the current axis modes).

Also @ukwebb, it looks like @OllisGit is using the Gcode Sending hook, which sort of gets rid of the plugin ordering problem since no gcode can be changed at that point.  However, for various subtle but important reasons, Octolapse currently tracks in the Queuing phase, and IS affected by plugin order.

FormerLurker, 2018-11-14:
@b-morgan, I think the key here is to look for the current max height where extrusion has happened.  This is seriously complicated by retract/detract.  Octolapse tracks retraction and extrusion too so that it knows what is a new layer and what is a zhop.

OllisGit, 2020-04-18:
Sorry, won't fixed....to complicated!

closed 2020-04-18 by OllisGit