OllisGit/OctoPrint-DisplayLayerProgress #50: Layer n of m should reset when a new print is started

b-morgan, 2018-05-25:
The Navbar display retains the display values of a print after the print is finished which is acceptable.

The Navbar display retains the display values of the previous print after starting a new print. This is not acceptable. 

When the Print button is pressed, the Navbar display should show values for the current print if known, and unknown values if not known yet.

OllisGit, 2018-05-26:
did you a reprint of the same model or do you select an other model?
i think i need to move the "resetValues"-code from selection to startprinting.

b-morgan, 2018-05-26:
This was with a reprint of the same model.

OllisGit, 2018-05-28:
It's fixed in the new Version 1.7.1.
Please test the bugfix.
Thx, in advance
Olli

closed 2018-10-21 by OllisGit