OllisGit/OctoPrint-DisplayLayerProgress #51: Feedrates should have a fixed width.

b-morgan, 2018-05-25:
The feedrates change alot and cause the Navbar to bounce around making it hard to read. I believe the feedrates need to have a fix display width.

If you don't want to expand the template / placeholder syntax to accept widths, I believe it is acceptable to start small and ratchet up so that if the initial feedrate is 100 then the width can be 3. If the next feedrate is 1000 then the width should be 4 and if the feedrate drops to 100, then the width should remain at 4.

Fanspeed could also suffer from this but since it is limited in range (0-100 or 0-255), a fixed width of 3 is reasonable.

jprochazka, 2018-07-10:
I am with you 100% b-morgan. The bouncing navbar is quite annoying when trying to do actions such as uploading a freshly sliced gcode file while a print is running. Personally I would be happy to see this feature removed or the information moved down into the state window where it belongs.

jprochazka, 2018-07-28:
As far as uninstalling the plugin I did the same a couple weeks ago. It is a shame I really liked the display progress feature and even the window displaying what the printer LCD is displaying but the bouncing nav bar is to much of a P.I.T.A. and IMO should be removed or moved to a separate plugin for those who feel the bouncing display is worth it to see the information displayed there...

@davidhbrown-uri idea pertaining to a fixed width probably would solve things. Better yet remove the nav bar information period. That or move it to the State area. I think I mentioned this in another issue but personally I feel you added so much to the plugin it has surpassed the display progress idea and added more than one would expect/what/need in a plugin entitled DisplayLayerProgress. The fix is trivial the problem is I am not familiar enough with Octoprint to make the changes and test them or else I would test and shoot over a pull request...

OllisGit, 2018-10-24:
Hi all,
I am back from a long vacation and build a new release V1.7.2.
In this version I added in default nav-bar pattern a span-element with a fixed width. So, now the annoying bouncing should be gone....maybe you need to tweak the pixel count, depending an your theme or browser.
Also, I removed the feedrate value from the default pattern, because not many users us such information.

Please try the new version and give me a feedback.

Thx, in advance
Olli

OllisGit, 2019-01-10:
no feedback -> closed

closed 2019-01-10 by OllisGit