OllisGit/OctoPrint-DisplayLayerProgress #58: Option to choose what to display when printing is finished

hnsantana, 2018-06-21:
When I send a G-code file with no layer information to the printer like this melody, for example:

M300 S2349 P53
M300 S0 P53
M300 S2349 P53
M300 S0 P53
M300 S2349 P53
M300 S0 P53
M300 S2349 P428
M300 S932 P428
M300 S2093 P428
M300 S2349 P107
M300 S0 P214
M300 S2093 P107
M300 S2349 P857

The message Print Done appears on the printer's display when the melody ends.

[image]

If layer markers (CURA format) are added in the previous example:

LAYER: 0
M300 S2349 P53
M300 S0 P53
LAYER: 1
M300 S2349 P53
M300 S0 P53
LAYER: 2
M300 S2349 P53
M300 S0 P53
LAYER: 3
M300 S2349 P428
M300 S932 P428
M300 S2093 P428
M300 S2349 P107
M300 S0 P214
; LAYER: 4
M300 S2093 P107
M300 S2349 P857

The final progress, as configured by the user, is shown on the printer's display. In my case, Job: 100% L=X / X appears on the display, where X is the value of the last layer.

[image]

So, can you add the option to choose what will be shown when you finish printing, such as a custom message?

Note: Octoprint has an option that allows you to display a custom message when printing is finished, but it is overwritten by the message that your plugin shows.

OllisGit, 2018-10-21:
Hi @hnsantana,
I am back from a long vacation and now I can support you with your issue.

You mean the GCode-Script feature "After print job completes", right?

I will take a look, what is the best way to send the "finish" message: via GCode-Feature or via the Plugin.

OllisGit, 2018-10-24:
Hi @hnsantana ,
I released a new Version 1.7.2. In this version I removed the "Print Done" message, because (as you mentioned) the complete message could be done with the default function.

Also I changed the default cura-pattern, because in your sample gcode above, there are a lot of errors. The ; is missing and also there are some spaces that are not recordnized by the parser.

Please try the new version and give me a feedback, if this solves your issue.

Thx, in advance
Olli

hnsantana, 2018-10-27:
Hi Mr. Olli,
In my personal opinion it is better to send the finish message through the G-Code.

I will test the new version.

Thank you for your attention.

closed 2019-01-10 by OllisGit