OllisGit/OctoPrint-DisplayLayerProgress #59: Max height always shown as 1.00 mm

EddyMI3d, 2018-07-08:
I'm running version 1.7.1 and in the status bar the total height is always shown as 1.00 mm, no matter what method I set to calculate the max height.
[image]

OllisGit, 2018-10-21:
Hallo @EddyMI3d ,
I am back from a long vacation and now I can support you with your issue.

Please provide a sample G-Code, so that I can reproduce the error.

BR
Olli

CDrawe, 2018-11-14:
Hello Olli,

I have nearly the same issue. It is stuck by showing 2,00 mm. 
The cause is quite clear. I'm using Octolapse with a lift and retract option. The lift option is exactly that 2 mm.
When I'm not using Octolapse the shown height is correct.

Any idea how to solve that issue?

Thanks in advance for your answer.

CDrawe, 2018-11-14:
Found a conversation about my problem in another issue...
Please close again.

closed 2019-01-10 by OllisGit