OllisGit/OctoPrint-DisplayLayerProgress #60: Too many features... maybe split things up?

jprochazka, 2018-07-10:
I installed the plugin expecting a layer count and now have more than I subscribed to. Think about splitting up your project and offering such things as the printer display and feed rate as separate plugins so we can pick and choose the information we wish to see instead of having to deal with more information than is needed just to see what the plugin is advertised to do which is to show what layer is being printed.

If you were to break the project into multiple plugins such as one offering the ability to display layer height and another to display what is displayed currently on the printer I would be all about downloading both. Feed rate however I could care less about and is making the UI unusable while a print is active.

tonejaco, 2018-07-10:
Huh? You configure what you want to see. If you only want layer progress, that’s all you configure.

jprochazka, 2018-07-10:
Oh crap did not see the option to remove the status bar spam. Still my opinion splitting features up into multiple plugins would be a plus being if I saw a plugin that was titled as displaying layer progress I would not expect the cool printer display feature which I am sure many would love to have without having to be forced to utilize the layer progress feature.

Just my two cents and why I will not close this issue myself. I will leave that the developer of the plugin to do so if he feels a feature rich plugin beyond the scope of simply displaying layer progress named as such is the direction he goes.

tonejaco, 2018-07-10:
The solution to the problem is perhaps to rename the plugin "Display Info". I think splitting it up is a mistake. Just adds extra overhead and then you have to go to multiple places to configure what you want to see. I like it the way it is but I do agree that the name can be misleading.

jprochazka, 2018-07-10:
My thought is layer count is cool, showing what is displayed by M117 to the display on the printer on the web page is cool. Placing height, federate, ect. is cool for some but I find no want or need to see this info. The bouncing Navbar displaying this information, at least for me, is worthless information causing the screen to bounce around makes me want to claw the eyes out of new born puppies when trying to interact with Octoprint.

Apparently someone loves this annoying feature. Make it a separate plugin and give back the plugin I once liked before this information was added causing a real annoyance while a print is in progress.

OllisGit, 2018-10-21:
Hi everybody,
I am back from a long vacation and now I can support you with your issue.

@jprochazka Thanks a lot for your critical opinion, I appreciate that.
Feedrate:
I totaly agree that I need to fix #51 a.s.a.p. This is really annoying. 
Also, I will remove the feedrate text from the **default** Navigationbar text, it's too long.

Spliting the Plugin:
I will not split up the plugin, because I takes more effort to maintain multiple plugins.
But I am thinking about the consequence, if I change the name. 
Maybe add a postfix like "DisplayLayerProgress and more Plugin". I am not sure right now.

Show/Hide Statusbar bug:
You mention that you uncheck the two **Statusbar** flags and the text appears in **Navbar**.
Was that a typo? Because the statusbar is on the left and the navbar is on top.
If you want to remove height/layer information from navbar you need to change the Navigationbar text template.
Or please so kind and create an other issue if the statusbar still shows hight/layer after you uncheck the flags in the settings (maybe a browser cache-problem, press Ctrl + F5 for reload).

BR
Olli

OllisGit, 2019-01-10:
no feedback -> closed

closed 2019-01-10 by OllisGit