OllisGit/OctoPrint-DisplayLayerProgress #61: [Feature Request] Tuning the delay between printer messages

PeggyFree, 2018-08-16:
Hello, thank you for this very useful plugin.
I've sightly customised the plugin by myself because the PNotify popup was hidding important information when stack at bottom left, so I moved it to the bottom right.

One thing I could'nt tune is the delay between two messages on the printer.  Currently each message (ETA/ETL/Percent/accuracy) is shown 5s or something like that. To be honnest I don't want to wait 15s to know when the print will end for instance. Is it possible to tune this? **The best solution would be to tune this delay per message** (for instance accuracy = 1s, % = 2s, ETA = 5s, ETL = 3s). But a unique value would do the trick. 

If you don't intend to enhance the plugin settings for this : where may I modify the source code to change this by myself?

OllisGit, 2018-10-21:
Hi @PeggyFree ,
I am back from a long vacation and now I can support you with your issue.

I am not sure if I understand your request correct, because there is no internal duration how long a message will displayed. 
The message in the popup will be visible since a new M117-message will be send to the printer.

Please give me more informations:
Olli

PeggyFree, 2018-10-21:
Hello, 

Thank you for coming back, I hope you enjoyed your journey. The point is how do we tune the frequency the M117 is sent to the printer? Currently 5s seems too long.
If this parameter can't be tweaked, is it possible to display ETL and Percent longer than ETA and accuracy? For instance send ETL and Percent twice, si that they would be visible 10s rather than 5s, so greater chance to see them when looking at the screen.

This is not an issue, but a improvement suggestion.

OllisGit, 2018-10-21:
hmmm....I think you are in the wrong Issue-Tracker. My plugin doesn't use the terms "ETL" and "ETA".
Take a look at "https://github.com/dattas/OctoPrint-DetailedProgress"
And here is the homepage text:

> By default, It rotates the display every 10 seconds with the following information:
> 
> - Percentage of the job’s progress
> - ETL - Estimated time left
> - ETA - Estimated time of completion

PeggyFree, 2018-10-24:
My bad, thank you very much for your help!

OllisGit, 2018-10-24:
No problem!

closed 2018-10-24 by OllisGit