OllisGit/OctoPrint-DisplayLayerProgress #64: conflict plugin ipOnConnect

ygtc, 2018-10-26:
Hi,

I noticed that this plugin has a conflict with ipOnConnect plugin.When screen display ip then we connect web interface of octo your plugin update itself like - "H: -/-  L: -/-"

Could you configure option only update status when printing? Bcs this is really annoying for me :)

Thank you for this plugin.Kepp up the good work!

OllisGit, 2019-01-10:
@ygtc 
Issue is fixed in V1.8.1.
There is a new option "Update Printer Display only while printing". If checked, the Plugin doesn't send Layer-Information to the display -> no conflict with Plugins that use M117 command before printing.
Sorry, for the long delay!

ygtc, 2019-01-12:
Yes, this issue is fixed with "Update Printer Display only while printing". :+1: 

Thank You Olli :+1: :smile:

closed 2019-01-12 by ygtc