OllisGit/OctoPrint-DisplayLayerProgress #67: Display Estimated Time Left

TechnoSwiss, 2018-12-09:
Would be nice to have Print Time Left displayed (as show in OctoPrint web page), well added in as one of the place holders. If I understand the data the plugin has access to correctly, when you do a _printer.get_current_data() the data is available as currentData["progress"]["printTimeLeft"]. The value could be "None" since OctoPrint waits a couple of layers to calculate Print Time Left, in which case it could either display "None" or maybe currentData["job"]["estimatedPrintTime"] with an * or something to denote it's the estimated time.

Thanks

OllisGit, 2019-01-10:
I like the idea, so I implemented it in a feature branch.
[image]
You can test the version by downloading via this URL: 
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/archive/feature/%2367_printtimeleft.zip

After install, just add [printtime_left] in the Printer-Pattern.
If you like it then I will merge this feature into a next major-release.

BR
Olli

TechnoSwiss, 2019-01-11:
Thanks @OllisGit that works great.

OllisGit, 2019-01-12:
Feature is now in Version 1.9.0

closed 2019-01-12 by OllisGit