OllisGit/OctoPrint-DisplayLayerProgress #69: Miscalculated progress and heigth

pandel, 2019-02-21:
Hey!

Many thanks for your plugin. Normally, it is working as expected. But now I have a very big print that seems to be totally miscalculated. I'll show you some pictures:

Print time left, Printed and percentage are obviously wrong (static text in German, sorry):
[image]

You can see from the screenshot, the print is in the upper third of the print, so I'd expect the plugin to say something about 75% done and the other values accordingly.
[image]

What can I do to find out, what is going on here?

Regards,
Holger

pandel, 2019-02-21:
Just for the record: I kept an eye on it today and there really seems to be a bug somewhere. That's what I get now and the print will be ready in a few minutes (layer count is correct):

[image]

OllisGit, 2019-03-15:
Well, I am using the "orig." progress-value from OctoPrint and not a calculated value related to Layers.

Please look at the following forum-question and also the explanation from Gina in the YT-Clip.
https://community.octoprint.org/t/how-does-octoprint-calculate-progress/1265

Maybe I should add a hint in the Settings-Dialog...or even better: add a new key like `[layer-percentage]`, so that the user could decide which value he wants to see (progress or layer percentage).

What do you thing?

pandel, 2019-03-18:
Very interesting thread, I understand better now. I don't think that it is your turn to change something at this point. I even think about trying out gcodestat. Most of the time estimates are pretty close to what the printer is doing actually. Only if the files are extremely large (complicated, large object or many objects) the estimates are way off. So it is Octoprint's problem, not yours.

Many thanks for taking time to answer!

Regards,
Holger

closed 2019-03-18 by pandel