OllisGit/OctoPrint-DisplayLayerProgress #70: Screen Flickers, too many M117 commands

jkkato, 2019-02-24:
Hello, I like your Plug in, but I find that it it causes the display to flicker on my Ender 3. I think it is also slowing down my print. Especially on the curves. Looking at the Terminal, every 2 to 4 lines sent is an M117 command, with the same text. Is this normal?

[LayerDisplayM117.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/2897654/LayerDisplayM117.txt)

JMagoA8, 2019-03-11:
Hi! I've got the same problem in my Anet A8. If I disable the plugin the printer works perfectly, but if the plugin is enabled, at some points of the printing, the LCD screen blinks fast and the printer slow down and stops one second, in some cases.
I really like this plugin, but this is a big problem that mess up the printing process. I hope there's a solution for that, otherwise I'll remove this faboulous plugin.

[DisplayLayerProgress Problem.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/2953519/DisplayLayerProgress.Problem.txt)

jkkato, 2019-03-11:
I really like this plugin too. But some prints, it causes problems with the surface finish, because of the pausing while the ldc flickers. Very random. 

I have not heard back from the programmer. I ended up disabling this plugin. For now at least.

JMagoA8, 2019-03-12:
Maybe this is the problem we have with our printers.

https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/37

Later I'll try to print something without the height information, to see if this is the problem with the plugin.

JMagoA8, 2019-03-12:
I have check again my Code and jkkato's Code, and the M117 command is sent every time the feedrate of G0 or G1 changes, not because the height information.
While the author of this plugin solves the problem, I'll disable it.

OllisGit, 2019-03-15:
@jkkato  @JMagoA8 
hmmm..my first analyse doesn't showed an issue while sending G0 or G1 to the printer. The layer information is not send twice. Only if the feedrate is changed and the feedrate-key is used in the printer-pattern.

Please send my your gcode, so that I can reproduce the behaviour.
IMPORTANT: Download the gcode from Octoprint-Page after you load it for printing, because maybe the "layer-preprocessing" has a bug.

jkkato, 2019-03-15:
This is the exact Gcode, downloaded from octoprint, that was running when I saved the terminal text file in my original post. (as best as I can tell).

[CE3_Base LM2596 V2.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/2973212/CE3_Base.LM2596.V2.zip)

JMagoA8, 2019-03-16:
Hi!
In the .zip there are 4 files:
1.- A .gcode generated by Cura.
2.- A .gcode downloaded from Octoprint after upload the first .gcode.
3.- A .txt file with the configuration in the plugin menu.
4.- A .txt file with the orders sent by Octoprint to the printer.

[Plugin.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/2973750/Plugin.zip)

Thanks.

OllisGit, 2019-03-17:
Current status of the analyse:
- I think I found the issue!! It was really hard to find, because I needed to enhance the logging and the error occurs not very often in my setup.
-> multiple events at the same time.

Sent to the printer:
```
  2019-03-17 10:26:15,338 SENT-HOOK: M117 8% L=8/66 [17164::comm.sending_thread]
* 2019-03-17 10:26:18,543 SENT-HOOK: M117 9% L=9/66 [17164::comm.sending_thread]
* 2019-03-17 10:26:18,546 SENT-HOOK: M117 9% L=9/66 [17164::comm.sending_thread]
  2019-03-17 10:26:22,380 SENT-HOOK: M117 9% L=10/66 [17164::comm.sending_thread]
```
Line 2 and 3 were equal. 

The following log shows the commands that were send to the printer queue (before sent to the printer):
```
  2019-03-17 10:26:15,611 SENDING-HOOK: M117 INDICATOR-Layer9 [17164::comm.sending_thread]
  2019-03-17 10:26:15,614 UPDATE DISPLAY: layerChanged [17164::comm.sending_thread]
  2019-03-17 10:26:15,624 UPDATE DISPLAY: progressChanged [6984::Thread-58]
  2019-03-17 10:26:15,641 SEND-COMMAND: M117 9% L=9/66 [17164::comm.sending_thread]
  2019-03-17 10:26:15,641 SEND-COMMAND: M117 9% L=9/66 [6984::Thread-58]
```
The "layer" is changed at time ..614 and "progress" is changed at time ..624 (in different Threads), but the two commands were send to the queue at the same time ..641

I think that is the reason for flickering. Depending of the size of the print-model and also the performance of the cpu/firmware it flickers more/less often.

At the moment I don't have a good idea to solve it. 
Maybe compare the last M117 command that was send to the printer and if the same command is send again -> drop it!
I will do some implementation tests...

JMagoA8, 2019-03-17:
Maybe it's a problem, but I don't think this is the problem that we have.
In the next code the M117 command is sent by Octoprint to the printer multiple times and none of them is because of progress change or a layer change, because the percentage is the same in the commands, and the Z height not change.
I think it's problem of the feedrate, but I don't know why, because in the plugin settings I don't use the feedrate key.
`Send: N210 M117 1% L=0/66*93
Recv: ok
Send: N211 G1 F1350 X118.706 Y101.17 E17.74198*11
Recv: ok
Send: N212 M117 1% L=0/66*95
Recv: ok
Send: N213 G0 F3600 X118.14 Y101.17*64
Recv: ok
Send: N214 M117 1% L=0/66*89
Recv: ok
Recv:  T:204.96 /205.00 B:50.00 /50.00 @:72 B@:55
Send: N215 G1 F1350 X118.829 Y101.858 E17.77436*60
Recv: ok
Send: N216 M117 1% L=0/66*91
Recv: ok
Send: N217 G0 F3600 X118.829 Y102.424*69
Recv: ok
Send: N218 M117 1% L=0/66*85
Recv: ok
Send: N219 G1 F1350 X117.574 Y101.17 E17.83337*4
Recv: ok
Send: N220 M117 1% L=0/66*94`

`Navigationbar:

Fan: <span style='color:orange;font-weight:bold;'>[fanspeed]</span><span style='padding-left:30px;'></span>
Progress: <span style='color:orange;font-weight:bold;'>[progress]%</span><span style='padding-left:30px;'></span>
Layer: <span style='color:orange;font-weight:bold;width:30px;'>[current_layer]/[total_layers]</span><span style='padding-left:30px;'></span>
Height: <span style='color:orange;font-weight:bold;width:30px;'>[current_height]/[total_height] mm</span><span style='padding-left:30px;'></span>

Printer Display:

[progress]% L=[current_layer]/[total_layers]`

OllisGit, 2019-03-17:
The feedrate was also my first choice for a possible bug, but (as you mentioned) if the feedrate-keys are not in the message pattern, than no M117-command is send to the printer.

The terminal-output doesn't tell the "full-story" ;-)
Reason: After detecting changes (Progress, Layer, Z-Height, ...) the M117-command is not send directly to the printer. It is queue in a printer-command-queue.
Depending how full the printer-command-queue is, the M117 commands is sent delayed to the printer.

The tests with droping M117-command looks promising. Hopefully I can release a new version today.

JMagoA8, 2019-03-17:
But in other parts of the code from terminal output, when feedrate not changes there isn't M117 commands:

> Send: N78 G1 X96.049 Y99.445 E3.39957*88
Recv: ok
Send: N79 G1 X96.168 Y98.892 E3.41838*91
Recv: ok
Send: N80 G1 X96.364 Y98.362 E3.43718*88
Recv: ok
Send: N81 G1 X96.632 Y97.865 E3.45596*94
Recv: ok
Send: N82 G1 X96.968 Y97.41 E3.47477*111
Recv: ok
Send: N83 G1 X97.365 Y97.007 E3.49358*94
Recv: ok
Send: N84 G0 F3600 X97.756 Y97.201*112
Recv: ok
Send: N85 M117 0% L=0/66*98
Recv: ok
Send: N86 G1 F1350 X98.198 Y96.898 E3.51141*57
Recv: ok
Send: N87 M117 0% L=0/66*96
Recv: ok
Send: N88 G1 X98.681 Y96.663 E3.52927*88
Recv: ok
Send: N89 G1 X99.192 Y96.503 E3.54708*93
Recv: ok
Send: N90 G1 X99.722 Y96.42 E3.56492*106
Recv: ok
Send: N91 G1 X100.009 Y96.409 E3.57448*105
Recv: ok
Send: N92 G1 X119.991 Y96.41 E4.23908*95
Recv:  T:204.64 /205.00 B:49.95 /50.00 @:79 B@:77
Recv: ok
Send: N93 G1 X120.526 Y96.45 E4.25692*90
Recv: ok
Send: N94 G1 X121.049 Y96.569 E4.27476*97
Recv: ok
Send: N95 G1 X121.548 Y96.764 E4.29258*111
Recv: ok
Send: N96 G1 X122.013 Y97.032 E4.31043*96
Recv: ok
Send: N97 G1 X122.433 Y97.365 E4.32826*110
Recv: ok
Send: N98 G1 X122.799 Y97.757 E4.3461*90
Recv: ok
Send: N99 G1 X123.103 Y98.199 E4.36394*111
Recv: ok
Send: N100 G1 X123.337 Y98.682 E4.38179*89
Recv: ok
Send: N101 G1 X123.497 Y99.194 E4.39963*86
Recv: ok

OllisGit, 2019-03-17:
@jkkato @JMagoA8 
I released a new Version (1.10.0). 
Please test and give a feedback.

Thx, in advance
Olli

jkkato, 2019-03-28:
I have been away for a while (spring break with family), but I am testing the first print with the update right now. Looking good so far, no slowing down during the print, no screen flicker and the terminal gcode looks normal.

OllisGit, 2019-03-31:
Thx for the feedback , I appreciate that.
Olli

closed 2019-03-31 by OllisGit