OllisGit/OctoPrint-DisplayLayerProgress #71: A button for prints sent directly to the uploads folder from slicers.

Vantiveman, 2019-03-19:
Please note. This is not an issue - just a request.
This plugin is so useful, but it does not work for me because I slice using S3D and send the gcode directly to the upload folder of my Octoprint over the network. This plugin apparently uses the "upload" functionality to work so it does not apply the M117 changes to prints sent directly from slicers (at least, that is my experience).
Would it be worth adding a button to the plugin (in the Octoprint gui) to call your code and apply the M117 changes to gcode that has not used the Octoprint "Upload" button?
Currently I have a workaround of some VB.NET code I created to add these M117 commands at each layer, so I can apply the changes to gcode already sent to the Octoprint uploads folder, but it might be useful to other users as well as me to add an extra button for this.

OllisGit, 2019-03-20:
I don't use S3D, but the most slicers has some kind of scripting-feature. So you can add the necessary indicator via the slicer itself. No need for extra VB.NET programm.
Just add e.g. `;LAYER:19 M117 INDICATOR-Layer19` for a Layer.
If my plugin read this statment, the layer-information is send to the printer-display.

So, at the moment I don't see that need for an extra button. 

Probably I could change the "enhancement-process" from "uploading" to "selecting"-event. This should also solve your problem.

What do you think?

BR
Olli

Vantiveman, 2019-03-20:
> <OctoPrint-DisplayLayerProgress@noreply.github.com>
> sent directly to the uploads folder from slicers. (#71)

Hi,

OK, I just googled and found that in the S3D layer change script it is
possible to use a "replace" command, so I will try {REPLACE "; layer" "M117
Layer"}

Don't make any changes - it would probably annoy other users if the event
was changed - but I will play with the S3D layer change script and see if I
can make it work.

 

Thanks for your response. I accept, and agree, that my suggestion is
probably unnecessary.

Al

<https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/71#issuec
I00SlO3ks5vYi8-gaJpZM4b8Iul> .
<https://github.com/notifications/beacon/ArlHK9GwDXbUcThZw1plxXJ-JD8xBAcAks5
vYi8-gaJpZM4b8Iul.gif> 

                                

                                

                                

                                

               

               

               

               

               

               

               

               

               

               

               

               

               

               

 
<http://emtracker.intelliverse.com:3009/Etracking/images/intelliversetracker
.png?toAdd=OllisGit/OctoPrint-DisplayLayerProgress&etrack=true&pluginid=404d
f69f7c300c8198fb9c519679aa04&messageid=16A897EDD5C747B4B39D3FB67D00F9B7&subs
criberid=62E8C5CC34AC45A9E05>

Vantiveman, 2019-03-20:
Hi again,

Just to confirm, adding a replace command in the S3D post-processing does
the trick. Problem solved.

The VB.Net program I mentioned is an all-purpose gcode program. I can open
gcode, process to add temp changes (for wood filament grain effect) and
pause commands for filament changes. Also now adds speed changes between
levels. Plus it visualises the gcode.

The latest thing I added that I think would make a good plugin is code to
make speed changes (Feed rate and flow rate) between two layers, especially
for printing things that have very fine detail on top. For example, between
layer 150 and layer 200 slow the speed by 80%. It then adds the correct
commands at calculated intervals between the layers entered. It means that a
high speed can be used for the lower levels but where the extra stability
and accuracy is needed it slows down.

If you think it's a good option for a plug-in - feel free. At the moment I
don't know enough about coding plug-ins.

All the best

Al 

 

<https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/71#issuec
<https://github.com/notifications/beacon/ArlHK9GwDXbUcThZw1plxXJ-JD8xBAcAks5

                                

                                

                                

                                

               

               

               

               

               

               

               

               

               

               

               

               

               

               

 
<http://emtracker.intelliverse.com:3009/Etracking/images/intelliversetracker
f69f7c300c8198fb9c519679aa04&messageid=8AEFD0FA205A4117A54124E062F20ECF&subs

OllisGit, 2019-03-20:
It is always a "battle/challange" to decide which feature should be implemented in OctoPrint and which one should be in the slicer-software.
The example you mentioned  (layer dependend settings) is in my opinion a slicer-feature and has nothing to do with "controlling" the printer.
The most slicing-software could do this out of the box (e.g. Cura) 

I found this for Simplify3d: https://www.simplify3d.com/support/articles/different-settings-for-different-regions-of-a-model/
Maybe that is what your are looking for.

C ya,
Olli

closed 2019-05-18 by OllisGit