OllisGit/OctoPrint-DisplayLayerProgress #72: Analyse layers without reuploading files

h4nc, 2019-04-10:
Coming from here:
https://community.octoprint.org/t/eta-etl-and-displayed-at-printer-lcd/8736

Would be nice if we don’t need to reupload files so that the layers can be analyzed.

Thanks!

OllisGit, 2020-01-02:
Hi @h4nc ,
behaviour is changed in the newest version 1.16.0.

Please test and give feedback!

h4nc, 2020-01-02:
Will do, but at the moment it tells me that 1.15 is the current version.

h4nc, 2020-01-06:
I'm afraid I can't test this anymore. None of my files is older 4 months. I guess restoring a backup did this. So it looks like I can't confirm that it works. Sry.

OllisGit, 2020-02-17:
It's not possible to do. The only option could be start manually the layer-analyse. This must be handled in #101

closed 2020-02-17 by OllisGit