OllisGit/OctoPrint-DisplayLayerProgress #73: ideaMaker Layer Indicator config line

Lord-Quake, 2019-04-13:
For those using ideaMaker as the slicer here is what you need to add to the Layer Indicator config to get it to display and count:

1		[;LAYER:([0-9]+).*] ideaMaker

OllisGit, 2019-04-28:
I added this information as a default settings for the next version.

OllisGit, 2019-05-18:
@Lord-Quake hmmm... I looked again at your suggested pattern, but the default cura-pattern should also work (only the expression):

```
;\s*LAYER:\s*([0-9]+).*   CURA
;LAYER:([0-9]+).* ideaMaker
```
The only difference is that before/behind LAYER optional whitespaces are allowed in CURA. So should also work for ideaMaker.

I tested with https://regex101.com/ but maybe I need a sliced gcode. 
If it doesn't work for you, please so kind and provide me a gcode-sample.

Thx, in advance
Olli

Lord-Quake, 2019-05-18:
Well, the Cura version as posted in the default code did not work.
That is why I posted the revised code which works with my ideamaker.

OllisGit, 2019-05-18:
Please attach a ideaMaker sample g-code file. So, that I can check the behaviour with my plugin.

Lord-Quake, 2019-05-18:
[20mm_Calibration_Box.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3194170/20mm_Calibration_Box.zip)

OllisGit, 2019-05-18:
@Lord-Quake  Strange, I could not reproduce the behaviour!
This is how I tested:
- uninstalled the current plugin-version
- restart
- install the latest version 1.10.0
- restart
- plugin settings / RESET SETTINGS
- drag and drop your file
- download the modified gcode from octoprint (see attachment)
[20mm_Calibration_Box_2.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3194193/20mm_Calibration_Box_2.zip)

--> On each ;LAYER -Comment there is a M117 GCODE -> everthing is fine!

Which Version of the plugin do you use?
Maybe there is an old expression in the settings, please press "RESET SETTINGS" and try again.

Lord-Quake, 2019-05-18:
My setup is a clean install I have version is 1.10.0 running.
I'll try again when I have the time.

OllisGit, 2019-08-17:
No feedback -> closed!

closed 2019-08-17 by OllisGit