OllisGit/OctoPrint-DisplayLayerProgress #74: Feature: Last layer time

agibson2, 2019-04-13:
I find myself printing things where I would like to estimate how long until my next pause to change the filament color.  For objects that the layers are pretty consistent, this works well for me but requires me to time how long the last layer took.

Any chance for a feature to show the time that the last completed layer took.  Optionally the last 5 layer average or something like that would be useful too but just having the last layer time would be helpful.

OllisGit, 2019-05-06:
@agibson2  It will be included in the next version.

OllisGit, 2019-05-18:
@agibson2 
Done. Version 1.11.0 is out.

Please test and give a feedback.
Thx, in advance
Olli

agibson2, 2019-05-18:
Thanks.  I missed your previous message announcement.  Thanks for this!  As soon as my current print is done (an hour or so), I will test it out.

agibson2, 2019-05-18:
It works perfectly.  Thanks.  I used the line below to display minutes and seconds and average.

`LayerTiming:<span sytle='display: inline-block;width:24px;'>[last_layer_duration]</span> , avg <span style='display: inline-block;width:24px;'>[average_layer_duration]</span>`

closed 2019-05-24 by OllisGit