OllisGit/OctoPrint-DisplayLayerProgress #75: Slic3r Pattern

crami, 2019-05-01:
This Layer config works for me when using Slic3r (Prusa Version)

count       [;BEFORE_LAYER_CHANGE]   Slic3r

OllisGit, 2019-05-03:
@crami  Thx, for the info! 
I will add this pattern in the next Version as a default-expression.

OllisGit, 2019-05-18:
@crami 
Done. Version 1.11.0 is out.

Please test and give a feedback.
Thx, in advance
Olli

Powerprobot, 2019-06-11:
With the expression tester, i make this:
;\s*move to next layer\s*\D([0-9]+).*

The Prusa Slicer 2 comes with this GCode:
G1 Z14.000 F7800.000 ; move to next layer (69)

The Tester explain me Group 1 = 69 ; but it doesn't work...

only if i add in "Prusa Slicer 2", "Printer Settings", "Custom Gcodes: before Layer change":
;BEFORE_LAYER_CHANGE

This works, but why not the first?

Greetings
Sven

OllisGit, 2019-06-23:
Hi @Powerprobot,

good to know that you found the solution for your problem.

Btw: I didn't tested the new "Prusa Slicer 2" yet. Question: is the comment 'move to next layer (xx)' the default setting? 
So, is it valid to add this expression to my plugin-settings to support "Prusa Slicer 2.0" out ouf the box?
BR
Olli

Powerprobot, 2019-06-23:
Hi @OllisGit  ,
no, the default is "no comments on GCODE". You must enable the option "Verbose G-Code" in "Printer Settings / Output Options". You can find it in the Expert View.

I will search for a solution for a default installation in few days....

Have a nice day
Sven

Powerprobot, 2019-06-23:
Here is a pattern to count layer but it isn't safe if you have any single Z-move in Startcode.
It is for slicers there move the Z-Axis in a single command.
Pattern:
`count	[^(G1 Z)([0-9]+)\W([0-9]+)\s[^XY]]		Any Slicer`

I hope it would help.

See Ya
Sven

OllisGit, 2019-07-17:
@Powerprobot 
Thx, for your effort. Maybe I or someone else need this info!

closed 2019-07-17 by OllisGit