OllisGit/OctoPrint-DisplayLayerProgress #76: Ignore negative extruder moves when calculating max height

ruffle-b, 2019-05-20:
The default settings for cura-lulzbot and my TAZ6 does this as part of the end of print job:

G91
G1 Z20 E-5 X-20 Y-20 F3000                ; lift up and retract even more filament

This means that DisplayLayerProgress shows 20mm as the height even if the model is lower than that.

If the plugin ignored negative extruder moves this would solve the problem.

OllisGit, 2019-05-24:
Hi @ruffle-b ,
which Mode do you use in your plugin-settings (Tab: Height)?
The default is  "Max Z-Value in G-Code".
Did you already try:  "Max Z-Value with Extrusion in Z-Layer"

So, if I understand you correctly the plugin shouldn't count hights where negativ extrusion is done.
If yes, I would try to change the "Max Z-Value with Extrusion in Z-Layer"-Mode.

It would be helpfull if you could provide a full gcode-file.

Thx, in advance
Olli

ruffle-b, 2019-06-03:
Hi,

I've got the plugin set to "Max Z-Value with Extrusion in Z-Layer" and for a 5mm test print the plugin shows 20mm height but the Octoprint GCode viewer shows 4.48mm (0.22 layer height with an 0.3mm first layer).

I've attached the gcode (extension renamed to .txt as github doesn't like .gcode) and a screenshot of the model loaded.
[LTAZ6A_20x20x5mm-Cube.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3247842/LTAZ6A_20x20x5mm-Cube.txt)
[image]

HTH

ruffle-b, 2019-06-03:
Here's a taller model that also shows 20mm in the Display Layer Progress Plugin but correctly(ish) in the gcode viewer.
[image]

ruffle-b, 2019-06-07:
Oh yeah.  But that's a move without extrusion isn't it?

The next move with extrusion is line 64 which would also be wrong if the model was < 2mm but it is an extrusion of 0. Should that count?

FWIW, here's a 65mm model showing a height of 20mm so there's something else going on (gcode attached as .txt).

This is all standard Cura-Lulzbot stuff so it won't just be me that seeing the incorrect model height.

[Phone Bracket Platform.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/3265066/Phone.Bracket.Platform.txt)
[image]

OllisGit, 2020-08-17:
Hi,
I dropped the whole "height-detection" in the newest release V1.23.0. Now the value is picked up from the OctoPrint MetaData.

Please test and give feedback.

Thx, in advance
Olli

closed 2020-09-26 by stale[bot]