OllisGit/OctoPrint-DisplayLayerProgress #77: [BUG] not correct layer nubers displayed on vase mode

njozsef, 2019-06-18:
HI developer

I use your plugin, great work.
Please check the MARLIN gcode (try dryrun plugin). gcode is vase mode.

Layer: 8814 / 128584
object only 150mm height and 0.1mm layers.
https://drive.google.com/open?id=1n5F_Sn91-8SgwUPkKeA3fuT91BmfBGqX
but LCD display is OK!
https://photos.app.goo.gl/ExA7Mk5F1rRim4L59

thankyou.

OllisGit, 2019-06-23:
@njozsef  sorry, but I can't reproduce the error.

[image]

I uploaded your gcode file and the plugin shows max 1507 layers .

* Please download the "Vase_Mode.gcode" directly from octoprint
* Open the file in a text-editor
* look for 'M117 INDICATOR-Layer' you should only find layer 1-1507

hmm... do you have other models in the filemanager that could match the layer-count of 128584?

OllisGit, 2019-10-01:
No feedback -> closed!

closed 2019-10-01 by OllisGit