OllisGit/OctoPrint-DisplayLayerProgress #78: How hard would it be to make the plugin mimic the same screen thats on the printer like say reprap screen

daimein2007, 2019-07-09:


daimein2007, 2019-07-09:
How hard would it be to make the plugin mimic the same screen that's on the printer like say RepRap screen? I would love any guidance on making the same acreen as my printer but inside ocotprint. I use this plugin now with the speech plugin I changed L= to layers= so she says it correctly but would be awesome to just have a clone of the machine's screen is that even possible. I know you have a lot of work in this I was just curious how big of a venture something like I suggested, would be.

OllisGit, 2019-07-12:
@daimein2007 you mean something like this ? 
[image]
So, you want see "everything" like XYZ-Position, Bed/Nozel Temp, Fan, .... and of course with the same pixel-graphics, right?

First question that come to my mind is, why should I display reduntent values, that are already shown in Octoprints-UI.

Short: It is possible, but I takes some effort to draw the animated graphics. If you want to start developing your own plugin I can support you.

BR
Olli

OllisGit, 2019-08-17:
No feedback -> closed!

closed 2019-08-17 by OllisGit