OllisGit/OctoPrint-DisplayLayerProgress #79: API for DisplayLayerProgress

TimonGaebelein, 2019-07-09:
Found this plugin somewhere mentioned in a thread back from May'18. You mentioned, that you may be able to supply all the data via an API Endpoint? Is this still on the roadmap? Would be super cool if this can be done, because I need the Fanspeed and Layer information via the API for a project! If you need any help let me know.

Thanks!

OllisGit, 2019-07-13:
@TimonGaebelein  I just realized that you created a Pull Request yesterday....thats bad for me, because I also created a solution in the last 2h and pushed that stuff as a new Plugin, see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress-API

So, I will take a more detailed look into you solution in the next days.

THANKS A LOT FOR THAT, because the new plugin is in a "rough-state". Feel free and give me a feedback on that.

Thx,
Olli

TimonGaebelein, 2019-07-13:
Yours is way more sophisticated. Although I have a few questions / improvements in mind which could be useful:

First of the echo call, I don't think it is something that will be useful for the Plugin, so it only adds additional load to the system. In general I like the idea of having multiple endpoints, but I, personally, think it only adds unnecessary load on the system (flask needs to check each Blueprint, so 3 false checks until it finally reaches the one, that retrieves the DisplayLayerProgress Plugin one.

My idea basically was to just add a single API endpoint to the existing plugin, so it'll just return the data, that is gathered by DisplayLayerProgress, but this might be fairly specific to my use case, because I don't need to know what the last Event was. A good idea could be to have slimmed down version in the main package and for everyone, who wants to get more information the API plugin could be useful.

And the last thing is, that the JSON return structure does not feel right. Octoprint normally consolidates things into sub-attributes, so you have multiple levels, which makes the code, that queries the API a lot easier to read. I've done that in my PR and I think it matches the JSON format a little better.

But you did great work! That API of yours is way better suited for extended use cases, but I'd like to see a really lightweight API in the main plugin, which just returns the data, that is collected by your awesome plugin. If you see that different feel free to close my PR and I'll switch to the API plugin, it will increase the load on the Pi though, which is something to keep in mind. If you're ok with having a default API on your main plugin feel free to merge my PR or request some additional changes, I'll implement them ASAP :)

TimonGaebelein, 2019-07-13:
Maybe to give you some contect: I'm currently working on a small dashboard / control center for Octoprint, which could be run on a 3 - 7 inch display directly attached to the raspberry (I really don't like those monochrome screens).

So I'd need the Fan value and LayerProgress from your plugin. The whole thing should be an electron app, which consumes a quite significant amount of the available ressources, therefore I dont really like the idea of having an additional package.

Here is a link if you want to have a quick look: https://github.com/TimonGaebelein/OctoprintDash

UnchartedBull, 2019-08-11:
Hi, any updates on this?

OllisGit, 2019-08-17:
@UnchartedBull  sorry for the delay,  but I was very busy the last weeks.

I created a new version which includes a new Endpoint:  /plugin/DisplayLayerProgress/**values**

I couldn't reuse your pull-request 100%, because I want to use the default-url for updating the display.
So, if you want to include the endpoint in your fancy looking dashboard, just add **/values** postfix.

Thx, for your contribution.
Olli

UnchartedBull, 2019-08-17:
Thank you man! Great idea to put it into a separate endpoint. Looks like it fully fulfills my needs. So this issue can be closed. Thanks again!

PR #80 (2019-08-17): API Endpoint
I've added a simple API to retrieve the data your plugin collects. I've tried to match the style of the Octoprint APIs in some way.

I think I've included everything that needs to be exposed via an API, I don't know what FeedrateG0 and FeedrateG1 are, though. If they are needed just let me know and I will add them.

This also closes #79.
closed 2019-08-17 by UnchartedBull