OllisGit/OctoPrint-DisplayLayerProgress #82: [MAXZ] Total Layer height on Cura 4.2.1 isn't working anymore

pseudex, 2019-08-18:
With the new Version of Cura (4.2.1) they changed the gcode in terms of Z movements.
This will only be done without extrusion.
Both options don't work for me. This is maybe because I'm using a double extruder which has relative G-code at the end.

this is the last Z movement while printing. The Z movement for next layer is done at the end of the actual one. Without Extrusion.
```
;MESH:NONMESH
G0 F600 X116.303 Y116.574 **Z84.36**
G0 F9000 X118.573 Y116.916
G0 X119.511 Y117.169
G0 X119.213 Y117.507
;TIME_ELAPSED:19307.906198
;LAYER:526
M117 INDICATOR-Layer526
```

I would suggest to read the MAXZ from the "header" which is put in by cura on default. To make it universal for more printers there could be an option to specify a reg expression for the max z line.
```
;FLAVOR:Marlin
;TIME:19310
;Filament used: 0m, 16.0813m
;Layer height: 0.16
;MINX:82.949
;MINY:102.45
;MINZ:0.2
;MAXX:152.051
;MAXY:132.551
**;MAXZ:84.36**
;POSTPROCESSED
;Generated with Cura_SteamEngine 4.2.1
```

OllisGit, 2019-08-19:
@pseudex  I release a new version [1.13.0 ](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.13.0) with your requested feature.

Please test it and give me a feedback.

Thx, in advance
Olli

pseudex, 2019-08-19:
wow. that was quick. when my 6hr print is done I will test it!
Thank you

pseudex, 2019-08-20:
[image]
works perfectly. thank you!

closed 2019-08-20 by pseudex