OllisGit/OctoPrint-DisplayLayerProgress #83: [FR] More observables in ViewModel

StefanCohen, 2019-08-27:
Hi,

I'm currently developing a Dashboard plugin for OctoPrint. You can follow my progress [here](https://community.octoprint.org/t/dashboard-plugin/11536/23) if you are interested:

I like your plugin and would like to use some of the print stats from DisplayLayerProgress when it is installed rather than re-inventing the wheel. I'm aware of both the API and event extensions and that will probably work. It would however be much simpler if the following properties where made observable in DisplaylayerprogressViewModel just as self.navBarMessage is now:

   'totalLayer'
   'currentLayer'
   'currentHeight'
   'totalHeightWithExtrusion'
   'feedrate'
   'feedrateG0'
   'fanspeed'
   'progress'
   'lastLayerDuration'
   'averageLayerDuration'
   'printTimeLeft'
   'printTimeLeftInSeconds'

Is that something you would consider adding?

OllisGit, 2019-08-28:
@StefanCohen short: No!

In my opinion the viewmodel is not "designed" for data-exchange between different  plugins.
As the name says, it includes data which is relevant for the view. Thats also the reason why i send the formatted message to the ui and not all the single properties.
I don't want "logic" in the UI ( maybe I am a little bit "old-school" ,-)

Another point is, I want the freedom to change my viewmodel any time I like, without receiving a lot of tickets why I changed this and that.

I had the same problem as you: I was looking for something like an Interface/Dependency-Concept, but this is not available. 
So I deceided to use the EventBus (V1.11.0+) to send data in a "fixed" format and every other plugin developer could grab that information when occurred.

StefanCohen, 2019-08-28:
That makes sense. I ended up using the events.

closed 2019-08-28 by StefanCohen