OllisGit/OctoPrint-DisplayLayerProgress #85: [FR] averageLayerDurationInSeconds and lastLayerDurationInSeconds

StefanCohen, 2019-09-27:
Hi again!

I have a humble request. Can we get averageLayerDuration and lastLayerDuration in unformatted form via the Event Notification and the REST API as:

- averageLayerDurationInSeconds
- lastLayerDurationInSeconds

This is similar to how timeLeft and timeLeftInSeconds works and would allow subscribers to use the values without having to guess how the user may have formatted the time strings.

Best Regards,
Stefan

OllisGit, 2019-09-29:
Hi @StefanCohen ,

the implementation of your request is done! 

**BUT** I only pushed the changes to the master branch and didn't release a new version.
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/archive/master.zip

**BECAUSE** the changes breaks the current behaviour 
current:```  "averageLayerDuration": "0h:01m:03s" ```
new: ``` "averageLayerDuration": 63 ```

I think if I release the new version you receive 10.000+ issues, because the Dashboard didn't show the formatted time any more.

Two possible options:
- synchronies our next releases
- you implement some kind of "corruption/backward layer" and I wait till your version is released

What do you think?

StefanCohen, 2019-09-29:
I think I was a bit unclear. What I meant was for lastLayerDuration and averageLayerDuration to remain exactly as before (with applied formatting) while the new fields lastLayerDurationInSeconds and averageLayerDurationInSeconds would provide the unformatted values. Like this:

`
{ 
   'totalLayer':'66',
   'currentLayer':'22',
   'currentHeight':'6.80',
   'totalHeightWithExtrusion':'20.0',
   'feedrate':'2700',
   'feedrateG0':'7200',
   'feedrateG1':'2700',
   'fanspeed':'100%',
   'progress':'28',
   'lastLayerDuration':'0h:00m:03s',
   'lastLayerDurationInSeconds':'3',
   'averageLayerDuration':'0h:00m:02s',
   'averageLayerDurationInSeconds':'2',
   'printTimeLeft':'2m3s',
   'printTimeLeftInSeconds':123
 }
`
This is as far as I can understand how you already have implemented printTimeLeft vs printTimeLeftInSeconds and would mean that you won't break backward compatibility and I can simply pick up the new values if they are provided and apply formatting.

OllisGit, 2019-09-29:
DAMN, I have no clue what I was thinking!

I will change the implementation in the next days...with a fresh mind ;-)

StefanCohen, 2019-09-29:
No problem at all.

Here's what I intend to do with the info once it is available:
[image]
I want to chart the layer times and layer average. 

Thanks!
//Stefan

OllisGit, 2019-09-30:
The looks really good! This board should be included in the offical OctoPrint ;-)
Small, compact and all informations in one spot.

I changed the output. 
It is not released, so if you want to test it use the above master-download link.

BR
Olli

StefanCohen, 2019-09-30:
Thanks. I will try it and see how it works for my scenario.

StefanCohen, 2019-09-30:
It works exactly as I had hoped. Thanks!

I will keep an eye out for the release where this is included and work on the graph in the meantime. 

//Stefan

StefanCohen, 2019-10-01:
I got the layerDuration graph to work. I have some styling to do but it is mostly ready now:

[image]

Thanks again!

OllisGit, 2019-10-01:
Fyi: New release is out!

closed 2019-10-11 by OllisGit