OllisGit/OctoPrint-DisplayLayerProgress #86: Layer information not displayed - cura slicer

RyanParle, 2019-09-27:
I recently installed this plugin to work with Navbar status & Dashboard plugins and have found that the layer information is not being displayed. In the status I am just getting **% -/-
I have looked at my .gcode files and they all contain the layer info in the format of ";LAYER:10"
Are there any additional steps to perform to read the layer information from Cura .gcode?

OllisGit, 2019-09-29:
After uploading the gcode file to octoprint, the plugin adds a M117 command behind the layer information.

Please download the gcode file (after uploading) from the octoprint-server (download-button right to the filename). The downloaded file should looks like this:

```
;LAYER:10
M117 INDICATOR-Layer10
```
With ```getting **% -/-``` you mean this is displayed in the dashboard, right?
Is the layer information displayed, after selecting the file, in the default sidebar?
If yes, it looks like a DashboardPlugin-Issue.

Do you already check the plugin-settings, maybe you deactivate the output.

RyanParle, 2019-09-30:
Ok, i didnt realise that your plugin makes changes to the gcode on import, that is where I was going wrong as the gcode files i tried were on the server before installing the plugin. all seems to work now.

closed 2019-10-01 by OllisGit