OllisGit/OctoPrint-DisplayLayerProgress #88: SD Card

novio8, 2019-10-02:
I use M23 command to upload to virtual SD and then M24 to print. It seems like Display layer progress does not work.
please advise

OllisGit, 2019-10-02:
Hi @novio8 ,
there is already an issue that is related to your scenario https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/72

Short: The gcode is changed during upload via WebUI. I am using `octoprint.filemanager.preprocessor`-Hook from Octoprint. This preprocessor is not executed when you directly (e.g. ssh, ftp) upload to your pi-filesystem (that what I understand from your description about M23/M24  or I didn't get it, please explain).

Do you have an option to use the Upload-API: http://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder
or
https://de.lmgtfy.com/?q=octoprint%20upload%20api
;-)

novio8, 2019-10-03:
Hi OllisGit, 
I use m23/24 for uploading gcode to Virtual SD and executed from there. I have Klipper installed and my Delta printer works extremely well. I would like Octoprint to show layers that are printed and how much time is left.  I don't have any programing skills so I could not understand how to use Upload_api from the link you'v provided.

OllisGit, 2019-10-04:
Hi @novio8 ,
hmmm...to be honest I am still didn't understand your printing workflow, so it is hard for me to give you an advice. 

As an example, my workflow looks like this:
- sclicing with cura
- in cura there is an octoprint plugin which sends the gcode to octoprint-server via REST-API
- then I can start the print in Octoprint.

M23 is for reading and not for uploading, that confused me (https://reprap.org/wiki/G-code#M23:_Select_SD_file)

Maybe you can explain your workflow in more details. How is octoprint in involved?

novio8, 2019-10-04:
sorry, maybe I did not explain it clearly. Klipper is Firmware that runs on RasberyPi or any other comp, linux, mac and win. It also installspart on 3D printers controller. All calculations are done by comp, printer controller just controlls I/O devices. Klipper uses Octoprint to communicate with 3d printer. I also use Cura to generate gcode, but not as plugin, I use it as stand alone version.

M23 is command that selects SD File and M24 starts or resumes SD print, just look at Marlin Gcode page.
I hoe that this helps

OllisGit, 2019-10-04:
hmm...not enough ;-)

- Who is generating the gcode?
- Who/How do you transfer the gcode to your printer? Target is the virtual SD-Card, right?
- Is the VSD-Card location in the OctoPrint-Setting: Upload-Folder?
- Do you select the gcode from Octoprint or via Printer-Display

I also use Cura as a Standalone app, but I installed an Octorprint Extension. Click on Marketplace and install "Octoprint Connection". 
After that you need to configure your printer in Cura to communicate to Octoprint. API-Key is needed. 
[image]
[image]

novio8, 2019-10-05:
1. Cura generates gcode
2. I upload file using Octoprint upload for example test.gcode
3. I select SD File using M23    m23 test.gcode
4. I start SD file using M24
so everything is done in Octoprint, except gcode generation

OllisGit, 2019-10-06:
okay, please attach two gcode files (as zip)
- the generated gcode, directly from cura
- and after uploading to OctoPrint (via the download button in the files menu)

if the second gcode file includes `M117 INDICATOR` then the plugin is working as expected.

To reproduce the error what task should I do? How can I create a virtual SD-Card?

BUT i think the problem is that OctoPrint process the sd located file not line by line, instead it is streaming directly to the printer --> I can't fix it!

novio8, 2019-10-08:
Hi,
I uploaded and then download gcode. It seems that plugin works, downloaded gcode
has new layer indicator lines: M117 INDICATOR-Layer0. 
Yes, you are probably right Octoprint is streaming gcode directly to the printer, because
GcodeViewer also does not work when I use my procedure M23/24.

OllisGit, 2019-10-11:
Sorry, can't help! Plugin is working as expected and I have no idea how to "manipulate" the gcode-stream.

closed 2019-10-11 by OllisGit

novio8, 2019-10-17:
thank you, I understand