OllisGit/OctoPrint-DisplayLayerProgress #89: Python 3 Compatibility

StefanCohen, 2019-10-15:
Hi!

Just a heads-up. I've successfully installed both DisplayLayerProgress and Dashboard under Python 3 using the OctoPrint devel branch and both starts without a problem. There is however some issue with the gcode parsing:

```
[2019-10-15 11:35:37,459] ERROR in app: Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/root/devel/oprint/lib/python3.6/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/devel/oprint/lib/python3.6/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/devel/oprint/lib/python3.6/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/devel/oprint/lib/python3.6/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/root/devel/oprint/lib/python3.6/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/devel/oprint/lib/python3.6/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/devel/OctoPrint/src/octoprint/server/util/flask.py", line 1294, in decorated_view
    return func(*args, **kwargs)
  File "/root/devel/oprint/lib/python3.6/site-packages/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/root/devel/OctoPrint/src/octoprint/server/api/files.py", line 371, in uploadGcodeFile
    display=canonFilename)
  File "/root/devel/OctoPrint/src/octoprint/filemanager/__init__.py", line 496, in add_file
    path_in_storage = self._storage(destination).add_file(path, file_object, links=links, printer_profile=printer_profile, allow_overwrite=allow_overwrite, display=display)
  File "/root/devel/OctoPrint/src/octoprint/filemanager/storage.py", line 765, in add_file
    file_object.save(file_path)
  File "/root/devel/OctoPrint/src/octoprint/filemanager/util.py", line 92, in save
    shutil.copyfileobj(source, dest)
  File "/root/devel/oprint/lib/python3.6/shutil.py", line 79, in copyfileobj
    buf = fsrc.read(length)
  File "/root/devel/OctoPrint/src/octoprint/filemanager/util.py", line 194, in read
    processed_line = self.process_line(line)
  File "/root/devel/oprint/lib/python3.6/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 108, in process_line
    line = self._checkLineForLayerComment(line, layerExpression)
  File "/root/devel/oprint/lib/python3.6/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 119, in _checkLineForLayerComment
    matched = pattern.match(line)
TypeError: cannot use a string pattern on a bytes-like object
2019-10-15 11:35:37,460 - tornado.access - ERROR - 500 POST /api/files/local (192.168.0.142) 10.58ms
```

This is just a heads-up. I don't know if or when you intend to attack support for OctoPrint 1.4 but I can start investigating now if you want to?

StefanCohen, 2019-10-15:
Update: I have a fix that makes this plugin Python 2 and 3 compatible. It is just two lines in the process_line function commented below:

```
    def process_line(self, line):
        line = line.decode('utf-8') #StefanCohen: Decode the input to a string
        for layerExpression in self._allLayerExpressions:
            origLine = line
            line = self._checkLineForLayerComment(line, layerExpression)
            if line is not origLine:
                # pattern matched, skip other expressions
                break
        # line = strip_comment(line).strip() DO NOT USE, because total-layer count disapears
        if not len(line):
            return None
        return bytes(str(line + "\r\n").encode("utf-8")) #StefanCohen: encode it back after processing and add a newline

```

I haven't tested it on Python 2.7 yet but this looks really promising and I haven't seen any other exceptions in Python3 so far.

Here's a job running:

[image]

OllisGit, 2019-10-16:
Hi @StefanCohen 
Thanks for the info, I appreciate it a lot.

Python3 is planned for X-Mas. How is your release plan for P3?
Do we need/should synchronize our P3 releases?

StefanCohen, 2019-10-16:
No problem. I haven't given my release plan a lot of thought. Before 1.4 is the deadline and I'm not really sure when Gina plan to release that except "before new-years". I just had an opportunity to try it out this week so I did and that's when I found that showstopper. That's actually the only thing I've found so far. I haven't changed anything else in DisplayLayerProgress nor the Dashboard, so I think this may be a quick fix to reach P3 compatibility. An easy way forward is would be for you to test my proposal now in 1.3/P2 to make sure  it doesn't break anything and then set __plugin_pythoncompat__ of your plugin to >=2.7,<4 That would allow me to test it more easily in P3 from now on. I can do the same on my end. What do you think about that?

OllisGit, 2019-12-21:
Hi @StefanCohen ,

I started switching to 1.4 and python 3.

I have some questions regarding your suggested solution:
*  why do you add additional linebreaks?
* why do you cast/convert to str and byte? my current solutions looks like this:
```
line = line.decode('utf-8')
...
line = line.encode('utf-8')
```

StefanCohen, 2019-12-21:
Hmm. That looks familiar. I tried that but never got it to work for some reason. 

Mvh,
Stefan Cohen

OllisGit, 2019-12-21:
Hi @StefanCohen 
the newest Version 1.15.0 is now running with OctoPrint 1.4.0RC3 and Python 3

Please test and give feedback
Thx, in advance,
Olli

OllisGit, 2020-01-30:
Latest version 1.17.0 is in "production" without issues related python3.

closed 2020-01-30 by OllisGit